opentrack/opentrack #572: Centered position sent to games.

olegiy, 2017-03-16:
Thus it must be?
[https://youtu.be/aC_BJE2RFiQ](url)

sthalik, 2017-03-17:
@olegiy turn off all relative translation options, including dynamic head and see if it still happens.

olegiy, 2017-03-17:
[https://youtu.be/bXaE_QZfZ4Q](url)

olegiy, 2017-03-17:
with Yaw, Pitch, Roll all good

sthalik, 2017-03-17:
Let's consider moving custom zero pose after all mapping and filtering. Having it done before relative translation is definitely incorrect here.

See the test build.

petsild, 2017-03-17:
Hi,
the shift of the center position of the pilot works fine, no motion to each other does not affect. Again great job thank you very much for adjustment. Pet

sthalik, 2017-03-17:
Thanks for confirmation. Fixed in 949d847ed9df6d2d2a7931f3f69014aa56a5ab7c.

closed 2017-03-18 by sthalik