opentrack/opentrack #574: jumping view

phantomwraith, 2017-03-16:
I have an persistent issue with my opentrack. With my head perfectly steady the view will suddenly jump, left then back for example. I tried another version of opentrack called trackhat opentrack and it has the same issue. Through the program it looks like all 3 dots jump together to a different point on the screen then jump back. I have videos showing this but i'm not sure how to share. I tried reinstalling and clearing regedit and it solved the issue for a couple months, but it's back now.

sthalik, 2017-03-16:
How far do they jump? A few pixels or more than 10% of the screen? If the latter, not enough USB bandwidth.

phantomwraith, 2017-03-17:
I think it's less than 10% of screen, if you give me your email I can send you the videos i have of the issue. Also i tried plugging my tracker to a powerbank, still get the same issue.

phantomwraith, 2017-03-17:
[image]

phantomwraith, 2017-03-17:
[image]

sthalik, 2017-03-17:
Make the camera gain show your room, for testing. Then see what happens. You'll probably get the left side wrap around to the right side.

It's not the clip. Plug your _camera_ to a different USB hub.

sthalik, 2017-03-17:
You can also reduce framerate but that's a crap solution. This is a hardware problem.

phantomwraith, 2017-03-17:
just to clarify i have a ps3 camera, and when running cl-eye test this jump NEVER occurs on it.

sthalik, 2017-03-17:
Change the framerate in cl-eye test.

phantomwraith, 2017-03-17:
Changed frame-rate to 60 and it's still occuring, should i still try a different usb port?

sthalik, 2017-03-17:
Yes. You probably have 2 controllers, one with blue and one with black slots. Choose the other one.

phantomwraith, 2017-03-17:
it's working PERFECTLY so far!!!!! it's late for me now so i cant really try it for long, but i think it's solved. thank you so much! i really appreciate the quick responses too :D

sthalik, 2017-03-17:
:1

closed 2017-03-17 by sthalik

sthalik, 2017-03-17:
Kropek woke me up at 5am. Kitty's so impatient. :cat2: