opentrack/opentrack #575: calibrator fix for PT and Aruco

sthalik, 2017-03-17:
I changed the calibrator so that if you stare at the same point, it won't register this point over and over again. This used to bias the calibrator making it hard to calibrate at all.

Please check the March 17 build -- <<http://opentrack.misaki.pl>>.

It's possible that we're collecting too few samples. Right now calibrator's "step" is 3 degrees for both yaw and pitch. This may be too little. For instance my run ended up with ~45 samples. We can technically add a lower step for pitch, given movements are more restricted.

Pinging the usual suspects. @MathijsG @Len62 @petsild :cat2:

sthalik, 2017-03-17:
You can check the sample count in debug output by:

```editbin.exe /SUBSYSTEM:CONSOLE opentrack.exe```

then a window will pop up and you'll get the sample count as calibration ends.

Len62, 2017-03-18:
Got a chance to try out this version. I was able to get a good calibration while moving very slowly, as opposed to needing to move fast before. Can't do the sample count as I don't have the required software.

sthalik, 2017-03-18:
Here's editbin for glorified sample count

[editbin.7z.remove-pdf.pdf](https://github.com/opentrack/opentrack/files/852458/editbin.7z.remove-pdf.pdf)

cheers

Len62, 2017-03-19:
Thanks. Can't get editbin to produce any output. Probably user error. So, put OT into debug mode, start OT, do calibration, then I've tried running editbin w/admin priv, either before, during, or after OT is running. I'm sure I'm just missing something here. Any particular directory that editbin should be run from?

sthalik, 2017-03-19:
You need to run editbin.exe from the command-line. Pass it `/SUBSYSTEM:CONSOLE opentrack.exe` assuming it's run from install directory.

Len62, 2017-03-19:
That's the way I've been running it, from a command prompt, passing arguments, in the install dir of OT. It appears a window pops-up after executing but it disappears before you can see any data.

MathijsG, 2017-03-19:
Works nice here :-). Don't notice a big difference since I already calibrate with just a few movements, because earlier you told me more inputs would actually make it less accurate.

sthalik, 2017-03-19:
@Len62 Given you've already done this, running opentrack for the next time will pop up the dialog window.

@MathijsG is slow calibration as good or better than before?

MathijsG, 2017-03-19:
Hard to say yet, will play with it further.

Len62, 2017-03-19:
No luck with editbin, but, I will say slow speed calibration is working fine. I did multiple calibrations, slow, fast, and in-between. Model position measurements are within 3-4 mm each time.

sthalik, 2017-03-20:
@Len62 note that now you can just stop for a few secs and continue afterward.

Do it as fast or slow in any moment you want, it doesn't matter to the code unless you break the invariant (no roll or position change) and have enough samples.

sthalik, 2017-03-20:
Just to make it clear -- you can move irregularly, stopping, making sudden and slow movements when you want.

sthalik, 2017-03-28:
How do you like the Accela change in the nightlies? It should help with staircrase-like diagonal movement. You probably need to set sensitivity lower.

Len62, 2017-03-30:
I did not change my Accela settings while setting up 2.3.1.5 and everything is fine. Still a little short on time but I'll try to test it with some adjusted Accela settings. Maybe I'll start with the default ini and re-enter data as opposed to carrying it over from a previous set-up.

sthalik, 2017-03-30:
@Len62 the changes aren't in 2.3.1.5. Get the snapshot at http://opentrack.misaki.pl.

Len62, 2017-03-30:
Just try the newest test version?

sthalik, 2017-03-30:
Yes. Try the one before steamvr if you encounter crashes. Tell me of the crashes if you find any though.

Len62, 2017-04-01:
Okay, had a chance to run the accela test build. Started with a default set-up. Did not run into any problems and no crashes. I left accela at default settings and movement was smooth. I can try playing with the accela settings to see what impact they may have, probably not until around this time tomorrow.

petsild, 2017-04-01:
The aforementioned is opentrack-test-20170326-accela-3dof.7z?

Len62, 2017-04-01:
Yes.

Len62, 2017-04-02:
Okay, I reduced all accela settings to their minimum and only had to adjust the pitch-roll-yaw settings to obtain nice smooth tracking.
[image]

petsild, 2017-04-03:
This weekend, I set up the new Accela filter and the axis of Yaw and Pitch movement is smooth but when I use the plus Z axis so a little bit of tearing the image. This is evident in the Rise of Flight or IL-2 Sturmovik Battle of Stalingrad.

The best settings:
[image]

Len62, 2017-04-04:
Interesting. I really only fly in IL2-BOS and have not come across any screen tearing. I'll have to load-up ROF and see if it shows up.

sthalik, 2017-04-05:
Tearing is lack of vsync and triple buffering. Nothing opentrack does can cause it by itself.

Len62, 2017-04-05:
I was thinking that myself. Graphics and game settings are the normal culprits if you have screen tearing.

sthalik, 2017-04-05:
It's a puzzle to me why adaptive vsync isn't commonly used... I really hate how low framerates get even lower with vsync. Either that, or use vsync but don't wait in lock-step for finalizing the frame.

MathijsG, 2017-04-05:
I will play with this around. Playing a lot of Rise of Flight currently.

Current settings with profile made earlier:
[image]

petsild, 2017-04-05:
I use the Z axis to the zoom which has resulted in greater sensitivity and demands on hw. I have to give the value of the gate valves of sensitivity of at least 0.4 mm otherwise it look extra nervous.

Managed better to eliminate the stair moves on the Z axis thanks to my new settings.
[image]

closed 2018-02-16 by sthalik