opentrack/opentrack #576: Yaw for mouse scroll? I would love to use this for work. (Non-game use)

FarSideHub, 2017-03-20:
I`m trying to work in an office with a computer but because of medical issues, I can't use a keyboard or mouse for even a few hours a day over weeks without getting pain that takes months to recover from. Ergonomic alternatives make very little difference.
This has limited my career to work that doesn't require lots of computer use. As voice recognition software has improved, I'm trying again, though.
I would really like mouse scroll to be implemented, or even hints on how to cobble something together from source, or a separate background process to receive yaw output and convert to scrolls or keypresses at threshholds would be amazing.

sthalik, 2017-03-20:
Hey,

You can access opentrack's data through the UDP socket output module. It sends 6 64-bit floating-point values in raw format. In Python the `struct` module exports `push` and `pack` functions that do the trick.  You could also use C# with `SendKeys` readily available. It's even easier to convert double-float to native double-float in that language.

As for how to emit mouse wheel events, see: <<https://social.msdn.microsoft.com/Forums/vstudio/en-US/ad2d3e85-7527-49dc-ac1c-a2b031b119ba/how-can-i-simulate-the-scroll-mouse?forum=vcgeneral>>.

The first comment is correct, additionally that's the exact way we're emitting mouse movement events.

I'm unsure whether to implement mouse wheel events in opentrack's mouse module. If you're able to understand how mapping works in opentrack, it should be pretty easy to implement and use.

Besides, I'm in talks about assistive mouse tech based on opentrack's LED tracker support. I'm told even quadriplegic people have click-devices separately, but I'm pushing clicking with the tracker alone on my own initiative.

As a fellow person with work-impairing disability I can only symphasize with your request.

sh

sthalik, 2017-03-20:
Move I think about it, I'll do it in regular opentrack. I have a paraplegic family member with spine broken fortunately just below upper chest, so the person can use their hands.

The person told me once about how kids were fucking around, and one made a movie-like snap-his-neck maneuver on one of the kids. So guy's paralyzed for life.

Also please be aware that branded releases of opentrack don't have additional functionality -- they're stripped down to be easier to configure, having defaults for the devices. Furthermore, I reserve moral rights and license the code under the same permissive ISC license as opentrack's unstable branch.

This project's been somewhat of a burden and responsibility to maintain on my own since 2013, but this actually gives me satisfaction.

You can expect the initial release prior to Tobii EyeX and HTC Vive support since it's less demanding and stressful to implement it.

sthalik, 2017-03-20:
Of course there's going to be excessive configuration options to make it suitable for multiple use-cases. I'm not assuming you @FarSideHub are of a particular disability or other.

FarSideHub, 2017-03-20:
Thanks for replying so fast and thoroughly. I've just skimmed through so
far. I'll have to read your replies carefully and use Google a bit to
understand properly.

I can really empathise with this kind of project being a burden. Don't let
anyone negative get to you, and don't feel bad about taking breaks as you
need them - you'll feel better and the result will be better in the long
run.
Sorry for the unsolicited advice. I don't like people burning themselves
out.

Disclaimer: What I have is mild, as disabilities go.
Doctors give different diagnoses, but I can use a computer for a whole day
with only moderate pain. It's when it continues a few hours a day over a
week or more that I start to lose sleep.
I'm also very thankful to be otherwise entirely able bodied. A family
member was recently in a wheelchair temporarily due to injury, and I'm
grateful for the insight gained pushing a chair.

I expect it would really help many otherwise abled people to have options
like this though, and it would be a real step towards opening up career
paths I'd given up on.

...A few minutes later and I've just read through a few issues comments on
github.  I'd like to say again that even though this is more satisfying,
please don't burn out!

I'll stick with your main version of opentrack. I <3 excessive
configuration options!

Again, thanks for being available and for the willingness to help, however
it turns out.

sthalik, 2017-03-20:
Thank you for your advice, I appreciate it. If you see my github profile, there are multiple strings of "white spots" with no activity whatsoever. I'm still at home with internet access, and there's always more stuff to do with opentrack. You'd imagine my having done something trivial once a week and a half sometimes, not several months of downtime like that. However it's getting better with the years.

I'd go for it as follows:

- actions like yaw, pitch, roll, zoom, slightly-move-left-or-right, will be configurable to either mode switches (see below) or functionalities (e.g. center, click when mapping is non-zero on axis, go back/forward as if on additional mouse buttons, toggle operation, emit keystoke sequence)
- it's going to be an "output" module
- action movements will only work if it's a decisive movement. minor movements are to be ignored, and optional spring effect will center toward user's long-term position since people shift in their chairs
- actions like positional shift or roll can switch modes, there should be an optional distinction between a short and long duration, bindable separately
- the user will still need to configure the software's input method (PT or aruco, or any other 6DOF tracker) as well as the assistive module
- we don't have to configure camera angle since Y position changes aren't practical as input anyway
- options like relative translation ought to be disabled by the user manually
- events like mode switches should show a notification in the tray area. I won't make sounds or graphic assets unless someone donates some under a permissive license.

Feel free to discuss my concept and make suggestions. Strangely enough, writing the preceding functionality is actually easy as hell compared to some stuff opentrack does and simmers take for granted.

sthalik, 2017-03-20:
Note to self: with the PT tracker and correct FOV and model size in millimeters, position is exact in centimeters, barring camera distortion. This gives a proper reference for action mapping.

FarSideHub, 2017-03-20:
Wow, that sounds amazing!
First thought is that it might be recommended to switch sides regularly for
commands - we left click often, and scroll down more often than up, which
could lead to postural problems in the long term. This might be more of an
issue for people who are paralysed and can't stretch when they feel
discomfort. There are different profiles for different games now, right? I
don't know whether changing profile could be assigned an action movement?
It may not be practical, but an option to invert every axis and some
discipline switching every fifteen minutes whilst getting used to the
system would ensure balance, and if it's done at first, the user would
become accustomed to it quite quickly. Of course, this could be achived
with different profiles as above, too.

I can imagine a customisable circle menu like this
<https://raw.githubusercontent.com/yandongCoder/circular-menu/master/examples/images/menu-sample.png>
but
without icons being very useful for changing settings or for people who
don't have enough mobility to do more than mouse x, mouse y and one action
to be able to mouse left, right, middle (for scroll) click and more. This,
being a gui, is very different from the rest of opentrack, though. I'm not
requesting this, just brainstorming.

Multiple inputs (a different feature request issue) could make the whole
thing super powerful - an ED tracker strapped to a knee or foot can do so
much more than a pedal! (I know this is a long while down the pipeline.)

I would like to make icons... maybe it could be my training to get used to
the new system! Baptism by fire! I would probably get better results by
searching out very generic icons on a good license and, if necessary,
asking permission to use them. I'm not much of a graphic designer. Just
using text should be fine until a couple of people are using it, though, I
guess.

I know enough about computers to understand that this should be a lot
easier than lots of what opentrack does in the background. Not easy, but
much more straightforward in terms of coding. Working out what options
users need could be tougher, maybe. Writing documentation - not the coding
type, but for non-coder end users of assistive tech - could also be more
work (another area I'd be up for helping with).

sthalik, 2017-03-21:
> I know enough about computers to understand that this should be a lot easier than lots of what opentrack does in the background. 

You nailed it. And yes, plenty of the stuff I've done over the years was challenging. Accela is actually the fifth version.

That radial menu you suggested looks actually fucking amazing. The good thing about them is that unlike regular submenus, they work with muscle memory.

sthalik, 2017-03-21:
I'm still in the talks, and will announce anything solid when able.

sthalik, 2017-03-21:
With regard to difficulty. I STILL don't understand Patrick's "coplanar POSIT" code, even with the paper and code hand in hand.

I don't _fully_ understand the pose resolution, the initial non-dynamic part, fully, despite all the years of self-taught math.

FarSideHub, 2017-03-22:
I can understand coplanar posit, having read Wikipedia, and can understand how I might do it manually, and that's where I'll stop unless I have special reason to pursue it further!

Glad you like radial menus. 
Another option might be a similar thing but purely horizontal. The user would, for example, briefly roll head right (around the z axis) to bring up the menu, then instead of the cursor moving, yaw highlights menu items and pitch like a nod selects the highlighted menu item and pitch like a nod in the opposite direction cancels out of the menu... 
Or moves to another row of menu items, and roll could be used to exit, or... So many configuration options...

FarSideHub, 2017-03-22:
I can understand coplanar posit *in a black box kind of a way - what goes in and what comes out.

sthalik, 2017-03-23:
I'll update you on the progress once it's over.

POSIT is a typical "hill climber" specialized for the PnP (points and perspective) problem. It's actually significantly more unstable than a generic math optimization hill climber used for the PnP problem. Look at the LED tracker's Z value and see the amount of noise compared to other DOF.

The sad part's that for coplanar models there's some tricky work required given the "multiple solutions" problem implied. Aruco's using a generic optimizer but when the flat surface's close to orthogonal it cycles between several solutions with a gap of 15-20 deg. There's no knowledge implied in a flat rectangular model that'd make one solution the correct one. Fortunately LED models have more useful shapes.

I still like the radial menu concept more than the alternatives. What I'm going to do is position menu items circularly on the screen, but skip art assets making them into a wheel. That still makes for intuitive submenus.

There's some issue where rolling affects other rotation components. I think we can manage taking into account how these changes are covariant.

sthalik, 2018-08-03:
The project is still active with TrackHat's support. In its present state it won't be discussed here, sorry.

closed 2018-08-03 by sthalik