opentrack/opentrack #577: Hotkeys on Debian 8 x64 ( KDE )

ljgdasfhk, 2017-03-20:
It is not possible to assign hot keys F1 to F12 and some others. They do not work when the window is minimized.
Only letters and numbers work well. But then the corresponding key does not work in other applications. For example, I can not write messages with this key.

sthalik, 2017-03-20:
Using the X11 protocol, `XGrabKey(3)` doesn't allow for passing through the keys. Unless there's some way to inject the key back via Xlib or xcb without receiving the same key again, no can do.

Global shortcuts are typically done via desktop environment API but we have to be DE-agnostic here. Not to mention Wayland's gonna come and last time I checked there was no protocol for key sniffing.

We _could_ do it through evdev and bypass X11 silliness that way, but that makes for Linux-specific code rather than X11-specific code. Note that evdev needs device node permissions.

But _really_ I only support Linux to run Valgrind on the codebase. Linux is an OS for developers. It's not ready for end users. It probably won't be for several more years at least, or ever.

ljgdasfhk, 2017-03-20:
Используя протокол X11, XGrabKey (3) не позволяет передавать ключи. Если нет возможности каким-либо способом вставить ключ обратно через Xlib или xcb, не получив тот же ключ еще раз, это не удастся сделать.

Глобальные сокращения обычно выполняются через API среды рабочего стола, но здесь мы должны быть DE-агностиками. Не говоря уже о том, что Уэйланд придет, и в прошлый раз я проверил, что нет протокола для поиска ключей.

Мы могли бы сделать это с помощью evdev и обойти X11 глупости таким образом, но это делает код, специфичный для Linux, а не специфичный для X11 код. Обратите внимание, что evdev требует прав доступа к узлу устройства.

Но на самом деле я поддерживаю Linux только для запуска Valgrind в кодовой базе. Linux - это ОС для разработчиков. Он не готов для конечных пользователей. Вероятно, это произойдет не раньше, чем через несколько лет.

> Note that evdev needs device node permissions.
I run opentrack as administrator. It's not a problem.

> Linux is an OS for developers. It's not ready for end users. It probably won't be for several more years at least, or ever.

Many people already play on Linux. There is already a Steam client for Linux and games in it for end users.
I hope that you change your mind. I believe in your strength and experience. Good luck to you. I look forward to solving this small problem. The rest of the program works perfectly for the KDE.

sthalik, 2017-04-06:
> I run opentrack as administrator. It's not a problem.

That relates to `plugdev` group or equivalent permissions. Try `evtest(1)`.

I use Plasma 5.9+ myself so that's a good test environment.

ljgdasfhk, 2017-04-07:
If I want to run the opentrack on behalf of the user, then I need to create a rule for the device. how to do it?

sthalik, 2017-04-07:
You need `/dev/uinput` rw rights.

miniskipper, 2017-05-15:
Commited a quickfix for hotkeys under X11, waiting for acceptance,
Every key can only be assigned once, so for example you can't use Meta+1 and Meta+Ctrl+1 (shouldn't be a problem, not that much assignable functions).

sthalik, 2017-05-30:
@miniskipper see if current unstable works for you, thanks.

miniskipper, 2017-05-31:
only works if window is focused

sthalik, 2017-05-31:
@miniskipper you were right. See if this change helps.

miniskipper, 2017-05-31:
@sthalik works :)

sthalik, 2017-05-31:
Great! Can you find a bug or a few in the present state? Consider symbol keys with different representation with shift, different modifiers, key repeat etc. I've had to make a table for the keys and this is always error-prone.

Also my pet peeve is shift+1 showing as "shift+!" on Linux. User needs to change it to "shift+1" on Windows manually.

ljgdasfhk, 2017-05-31:
Now hotkeys work fine (Debian 8 x64)

sthalik, 2017-06-01:
@miniskipper @ljgdasfhk :+1:

sthalik, 2017-06-06:
I'm reopening the issue since `shift+1` shows as `shift+!` on Linux. There's a way to fix it with keycodes and keysyms. Hence I'm hijacking the issue.

closed 2018-02-25 by sthalik