opentrack/opentrack #579: Compilation on Debian 8 x64 gcc 4.9.2 Qt 5.8

ljgdasfhk, 2017-03-21:
`[ 45%] Building CXX object options/CMakeFiles/opentrack-options.dir/bundle.cpp.o
In file included from /home/someone/opentrack-opentrack-2.3.1.5/compat/util.hpp:3:0,
                 from /home/someone/opentrack-opentrack-2.3.1.5/options/group.hpp:4,
                 from /home/someone/opentrack-opentrack-2.3.1.5/options/bundle.hpp:11,
                 from /home/someone/opentrack-opentrack-2.3.1.5/options/bundle.cpp:9:
/home/someone/opentrack-opentrack-2.3.1.5/compat/ndebug-guard.hpp:2:5: error: #error "NDEBUG defined, don't define it"
 #   error "NDEBUG defined, don't define it"
     ^
In file included from /home/someone/opentrack-opentrack-2.3.1.5/compat/run-in-thread.hpp:3:0,
                 from /home/someone/opentrack-opentrack-2.3.1.5/compat/util.hpp:4,
                 from /home/someone/opentrack-opentrack-2.3.1.5/options/group.hpp:4,
                 from /home/someone/opentrack-opentrack-2.3.1.5/options/bundle.hpp:11,
                 from /home/someone/opentrack-opentrack-2.3.1.5/options/bundle.cpp:9:
/home/someone/opentrack-opentrack-2.3.1.5/compat/ndebug-guard.hpp:2:5: error: #error "NDEBUG defined, don't define it"
 #   error "NDEBUG defined, don't define it"
     ^
options/CMakeFiles/opentrack-options.dir/build.make:106: ошибка выполнения рецепта для цели «options/CMakeFiles/opentrack-options.dir/bundle.cpp.o»
make[2]: *** [options/CMakeFiles/opentrack-options.dir/bundle.cpp.o] Ошибка 1
CMakeFiles/Makefile2:1230: ошибка выполнения рецепта для цели «options/CMakeFiles/opentrack-options.dir/all»
make[1]: *** [options/CMakeFiles/opentrack-options.dir/all] Ошибка 2
Makefile:127: ошибка выполнения рецепта для цели «all»
make: *** [all] Ошибка 2`

Then I commented out the line
`#ifdef NDEBUG
#   //error "NDEBUG defined, don't define it"
#endif

#include <cassert>
`
as before
And then everything is fine

sthalik, 2017-03-22:
Remove `-DNDEBUG` from `CMAKE_C_FLAGS_RELEASE` and `CMAKE_CXX_FLAGS_RELEASE`.

Please quote code or compiler output using

> three title-apostrophes
> text here
> more text
> three title-apostrophes

sthalik, 2017-03-28:
We now remove -DNDEBUG implicitly.

closed 2017-03-28 by sthalik