opentrack/opentrack #580: No Russian interface ( Debian 8 x64 gcc 4.9.2 Qt 5.8 )

ljgdasfhk, 2017-03-21:
The interface did not switch to Russian
But in 2.3.1.4 switched

sthalik, 2017-03-22:
Probably your i18n files are empty. That sometimes happens on heavy changes to a file or two.

checkout unstable. `make i18n`. Then `git checkout -f */lang/*.ts`. Then `make install`.

sthalik, 2017-03-22:
[image]
> ananke /home/sthalik/dev/opentrack/build % DISPLAY=:0 env {LANGUAGE,LANG}=ru_RU.UTF-8 install/bin/opentrack

sthalik, 2017-03-22:
```
ananke /home/sthalik/dev/opentrack/build % DISPLAY=:0 env {LANGUAGE,LANG}=ru_RU.UTF-8 install/bin/opentrack
libEGL warning: DRI2: failed to authenticate
qxt-mini: adding event filter
^C
```

[image]

sthalik, 2017-03-22:
I confirmed i18n `lupdate` and `lrelease` work on Linux.

sthalik, 2017-03-28:
@ljgdasfhk I simplified the translation generation code. Also try running `make i18n` manually.

closed 2017-04-20 by sthalik