opentrack/opentrack #581: Mappings not saved

lumptom, 2017-03-22:
Hi, I am trying to use newest build (2.3.1.5) and my mappings are not saved in my profile .ini file (using custom, not default). Once I set all axes and close Mapping window and open it again, curves are in default again. There is no Mapping value written in profile ini.

sthalik, 2017-03-22:
are any other settings from the mapping window saved?

lumptom, 2017-03-22:
i think no ... thisi is whole content of my .ini

```
[migrations]
last-migration-at=20160917_00~

[opentrack-ui]
button-center=39
compensate-translation=false
button-center-alt=40
compensate-translation-disable-source-yaw=false
compensate-translation-disable-x-axis=false
neck-depth=0
guid-center={CF9E0470-D041-11E6-8002-444553540000}
guid-center-alt={CF9E0470-D041-11E6-8002-444553540000}

[tracker-pt]
active-model-panel=1
camera-fps=75
camera-name=PS3Eye Camera
cap-x=100
cap-z=70
init-phase-timeout=100
min-point-size=3
```

lumptom, 2017-03-27:
any result from your investigation? 
do you recognize bug? 
if so, when do you estimate will be fixed?

sthalik, 2017-03-28:
I don't see any bugs. The next step is to provide you with a debug version to understand what's happening. I haven't yet done the work to give you that version.

I haven't seen this happening for other users either and that's interesting in itself. But that kinds of bugs are the worst kind. It's either one of these critical hard-to-find bugs or user error.

Bear with me for now. It's a hobby project and there's already a fair amount of pressure stemming from it.

lumptom, 2017-03-28:
Of course, no problem, Although setting up curves everytime I use opentrack is uncomfortable, it is not terrible, Maybe, as a workarround for me, you could list here sample of curve settings stanza in .ini file, I could try to customize it and put inside my .ini manually ....

Anyway I am ready to do whatever needed to help you improve this nice app. Feel free to contact me by mail once you will be ready with debug version.

Thanks, Lumptom

MathijsG, 2017-03-28:
Did you change anything reagarding your documents profile or your settings location? Started opentrack as administrator maybe? Here I don't have any problems with the newest version.

sthalik, 2017-03-28:
@MathijsG He can change the other settings.

@lumptom these are floating-point values in binary. Without a tool you won't be able to read or write them. But consider:

```
[spline-X]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@^\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[alt-spline-X]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@>\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[spline-Y]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@f\x80\0\0\0\0\0@I\0\0\0\0\0\0)

[alt-spline-Y]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@>\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@I\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[alt-spline-Z]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@>\0\0\0\0\0\0@R\xc0\0\0\0\0\0)

[spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@9\0\0\0\0\0\0@f\x80\0\0\0\0\0)

[alt-spline-yaw]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x2@ \0\0\0\0\0\0@\\\xc0\0\0\0\0\0@0\0\0\0\0\0\0@`\x90\0\0\0\0\0)

[spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@.\0\0\0\0\0\0@V\x80\0\0\0\0\0)

[alt-spline-pitch]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\x1@.\0\0\0\0\0\0@Q\x80\0\0\0\0\0)

[spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\0)

[alt-spline-roll]
points=@Variant(\0\0\0\x7f\0\0\0\xfQList<QPointF>\0\0\0\0\0)
```

lumptom, 2017-03-29:
@sthalik good news .... once I put your lines into my .ini and then change anything, new mappings were saved well. It seems like mappings were not saved before due to appropriate lines were not present in ini file. Maybe this could help you to focus on right part of code, which should create initial mapping lines in .ini file.

Anyway, my problem was solved and you have workarround for a while.

Many thanks,
Lumptom

sthalik, 2017-05-04:
This is very scary stuff. We use the same bits of code for managing all of the software's options. It has no right to fail like that. I've reread the code and there's nothing dubious going on. Perhaps I'll get a reproducible test-case another time.

In any case it's good it works for you.

stevecooperorg, 2020-10-27:
I _thought_ I had this issue and it was a user error problem! I was filling in the mappings, then closing the window with 'x' -- I guess I was expecting a save but it functioned like the cancel button. A dirty flag and an 'are you sure you want to discard changes?' would have set me straight 👍 

Or you'll just need to get a better class of user :)