opentrack/opentrack #583: [Request] Allow mouse shortcuts in Global Shortcuts

Roundlay, 2017-03-29:
I'd like to be able to toggle tracking on and off based on mouse input, rather than keyboard input. For instance: being able to toggle tracking off when clicking the right mouse button, then back on again with the same button. This would help with tracking in games that require you to aim down sights.

sthalik, 2017-03-29:
Given the infrastructure we already have it's easy to support. Give it some time.

Roundlay, 2017-03-29:
Great to hear! Thanks for the response.

samtheeagle, 2017-05-07:
Another vote for this item. I'm really impressed with how opentrack has improved, and I'm keen to move to it from FaceTrackNoIR, but the lack of mouse button binding is awkward. Generally I want to freeze the view so that I can manipulate controls in DCS or P3D, and to do that I have a hand on my mouse, not my joystick.

Looking forward to seeing this option available. Thanks.

sthalik, 2017-05-07:
Hey,

The proposed features aren't voted upon. I've already agreed to implement this in the indeterminate future.  See the sheer amount of open issues with the "feature" label on them. Prodding on the issue tracker won't help any.

You can likely use some third-party software to map the mouse key to a keystroke for the time being.

samtheeagle, 2017-05-07:
Ohhhhhh kaaaayyyyy then. I'll just be quiet. Sorry to take up a few seconds of your time.

weeeee, 2022-01-13:
I would like to see this happen

sthalik, 2022-01-13:
What's your use case anyway, is it the 4th and 5th button?

For more common buttons, they're still going to be passed to the game in addition to being handled by opentrack, so it's not too useful.

weeeee, 2022-01-13:
> What's your use case anyway, is it the 4th and 5th button?
> 
> For more common buttons, they're still going to be passed to the game in addition to being handled by opentrack, so it's not too useful.

When starting A10 cold and dark I want to use 4th and 5th mouse buttons, for Flashlight (LAlt + L), and Opentrack Toggle (keep looking at same spot until next toggle keypress)

When I'm using the flashlight I'm almost always search for switches and dials with the mouse often when pressing switches and dials I want to use toggle to pause headtracking for precise interaction, makes sense to have flashlight and toggle on mouse, as I am holding the mouse at this time and not hotas.

Also useful while in flight, eg interaction with TAD and other switches, I often need to toggle headtracking for precise button presses etc especially dials which require prolonged accurate interaction

I'm going to try and use autohotkey to do this for now.

sthalik, 2022-01-25:
The initial implementation doesn't take keyboard modifiers into account. So shift+mouse4 is the same as mouse4 or ctrl+mouse4. It should help people playing GTA and other action games through mouse emulation. Is there any use-case where modifiers for mouse shortcuts make sense?

https://www.dropbox.com/s/dy77gvym3s9dic6/opentrack-master-20220126_01.7z

Please report bugs in this build before the next opentrack release.

sthalik, 2022-01-26:
Closing, Please report bugs here or in the announcement discussion thread.

closed 2022-01-26 by sthalik