opentrack/opentrack #585: create your own game up load to opentrack

daviddb1951, 2017-03-31:
Hi,
I want to up load my game to opentrack.
I create a video game and built the 3D virtual character using Blender.org.
I reduce the polys and compressed it on save as to file.
I converted the obj file to a text file and using notepad to a ini file.
Questions
What is the maximum data that can be used?
How can I reduce the size to meet the requirement?
Am I missing a step to get this working with success?
Any suggestions are welcome.
Thank You

sthalik, 2017-04-06:
Are you asking for head tracking support?

Implement either freetrack (source code is in `freetrackclient/`) or npclient (in `contrib/trackir-client`). The latter's the actual client part, already implemented.

daviddb1951, 2017-04-06:
April 5,2017
 
Hi Stanislaw,
 
Thank you for responding to my email.
Yes,that is correct on the head tracking support to implement a game I programmed myself.
Installed it on open track.
I will use the information you have sent me.
Thank You,
David Brown

closed 2017-12-02 by sthalik