opentrack/opentrack #586: std::make_unique compile error. Is C++14 required?

sloth1, 2017-04-03:
I'm trying to build from source on x86_64 Linux. gcc version 4.8.5.

opentrack/gui/main-window.cpp:689:20: error: ‘make_unique’ is not a member of ‘std’
             tray = std::make_unique<QSystemTrayIcon>(this);

sthalik, 2017-04-03:
It should pass `-std=c++14` to the cxx compiler on its own. See the verbose invocation.

Some of the older GNU version have `make_unique` missing, I'm aware of that.

Either update the compiler or in the worst-case scenario, replace this invocation with `tray = std::unique_ptr<QSystemTrayIcon>(new QSystemTrayIcon(this));`

sloth1, 2017-04-03:
I found it was set to C++11 in cmake/linux.cmake. I changed it to c++14 and now I can compile.

closed 2017-04-03 by sloth1

sthalik, 2017-04-06:
For the record, `opentrack-platform.cmake` now deals with it globally

```
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_DEFAULT 14)
set(CMAKE_CXX_STANDARD_REQUIRED TRUE)
```