opentrack/opentrack #588: The image is reversed and double

dimdya39, 2017-04-05:
[image]
[image]

-In other applications the image is normal...
help please

sthalik, 2017-04-05:
Your USB hub can't handle the bandwidth. Switch the hub or reduce the framerate.

dimdya39, 2017-04-06:
mmm... But, the reduction in the frame rate, does not remove the problem!

I'll try to update the drivers of the motherboard......

dimdya39, 2017-04-06:
[image]
Update drivers did not help.
Why does the camera work normally in the CL-Eye Test?

sthalik, 2017-04-07:
We use different code than "Amcap" on which CL-Eye Test is based. In fact I have work ongoing work to import Amcap into opentrack, but it's a couple thousand lines of code (!). The entire Amcap project is closer to 10 thousand lines, half of opentrack. It's a lot of work to import it, and it's gonna introduce bugs for a while after it's imported. Also lack of Amcap is the reason we can't access the PS3 Eye camera dialog right now.

The corruption doesn't look like broken stride. It looks like a duplicated halves of a repeated image.

My recommended solution for webcam stuff is always to always install K-Lite codec pack Mega right after installing Windows, and not touching codecs at all thereafter.

Barring that, you may make it work using "ManyCam" and passing the real camera through it. You'll be able to select manycam from the camera list then. The watermark's removable in options and doesn't require registration. Older versions of ManyCam like 3.x are less annoying than the newer, bloated ones.

Please be aware that the problem affects way less than one percent of users, and finishing Amcap work doesn't seem so urgent. If only the project wasn't so badly written as to consist of ten thousand lines! Our present webcam code is a couple hundred.

sthalik, 2017-04-07:
In fact, look up "amcap" at http://opentrack.misaki.pl. I don't remember if that version shows received frame timestamps and size in bytes, or does it display the video preview in the interface. Either way it's WIP. But you'll be able to access camera options through that version, as a proof of concept, something that opentrack never did for PS3 Eye.

dimdya39, 2017-04-07:
Thanks for the answer! 
[image]
[image]

I will try to update the K-Lite codecs.
"ManyCam" - most likely will not be able to give a high frame rate.
In this case, P33 lose all meaning.

sthalik, 2017-04-07:
Consider installing Windows in a VM and redirecting the USB device to it. Then see whether the frame is right. If it is, it's a codec issue.

dimdya39, 2017-04-07:
На втором компьютере все нормально.
Кодеки переустановил, но проблема осталась.:(

sthalik, 2017-04-08:
I'm not skilled enough with codecs to help you, sorry. Everytime I had problems, had to reinstall Windows on a fresh partition. Maybe someone else knows how to refresh them forcefully.

petsild, 2017-04-09:
You are using the latest drivers?
https://onedrive.live.com/?cid=C2CF2FB58A81CF4F&id=C2CF2FB58A81CF4F%21107&parId=root&action=locate

dimdya39, 2017-04-09:
Yes, the driver version is the same.

Removed the KL-codec - and, .. reinstalled CL-Eye-Driver
The camera has earned!
Thank you so much!

closed 2017-05-04 by sthalik