opentrack/opentrack #590: Breaks FlightGear Display

rayinny, 2017-04-06:
I am new to OpenTrack and FlightGear.  This is my first TEST try at integrating them and it is failing with the following results:

-  THE FLIGHTGEAR DISPLAY IS IN FACT FOLLOWING THE HEAD TRACKING IN OPENTRACK.  That's working.

-  But _as soon as OpenTrack starts tracking, the display in FlightGear breaks_ as follows:
o  **In the cockpit view:**
    •  The cockpit view actually displays an outside view instead, where I'm looking at the front 
         wheel from beneath the plane.
      •  _Parts of the plane display are missing_ so I can look up from under the plane and 
          see the sky through gaps that shouldn't be there.
      •	 Other parts seem stretched out or incomplete.
      •	 In fact _the display is simply hardly recognizable_.  _It is broken_.

   o  **In the other views:**
Most of the other views allowed by FlightGear are similarly broken as the cockpit view.  And those that are not broken don't track with the head tracking in OpenTrack AS THEY SHOULD.

I am unqualified to judge if this is an OpenTrack or FlightGear problem so I submit it here for you to let us know which.

My setup is as follows:

-  For FlightGear:
o  I am using _FlightGear 2017.1.2_.
o  In the startup screen, under the Settings tab, in the "Additional options" section, I added code:
       _fgfs --generic=socket,in,25,localhost,5542,udp,headtracker_
o  I copied file _"headtracker.xml"_ from folder 
       _"{Program Files}\opentrack\doc\contrib\FlightGear\Protocol"_ to folder
       _"{Program Files}\FlightGear 2017.1.2\data\Protocol"_.
o  NOTE: It is amazing that the instructions to link FlightGear are in file _readme.txt_ buried in folder _"{Program Files}\opentrack\doc\contrib\FlightGear\”_ and I could not find them anywhere else.  I found it by luck.  YOU GUYS NEED A WEB SITE!!!  Equally amazing, I guess, is that no one on the Web seems to have published step-by-step instructions for getting OpenTrack to work with FlightGear.  That’s unusual these days.  Maybe FlightGear or OpenTrack is not as popular as I would like.

-  For OpenTrack:
o  I guess I’m using _OpenTrack 2.3.1.5_ since I downloaded it recently, but I can't be sure since again amazingly OpenTrack doesn't have an "About" menu describing the product and crediting the developers.
o  My configurations are:
    •  Input: _Aruco_ -- Paper marker tracker.
         Note: Paper marker is my choice because I'm just doing initial testing so have not yet invested in hardware.  
Note: I displayed the Aruco image on my cell phone and moved that around rather than print the graphic.
    •  Output: _FlightGear_.
    •  Filter: _Acella_.
    •  All other settings are the _default_.

With this setup, tracking is actually poor but that is not the concern in this test.  Just getting the system to work at all is.  Preventing the FlightGear display from breaking is.  Refining tracking is for later.

sthalik, 2017-04-07:
The cockpit view actually displays an outside view instead, where I'm looking at the front
wheel from beneath the plane.

Doesn't it move to the seat position after you center? Use the keybinding and optionally select "center on startup".

rayinny, 2017-04-07:
I think you misunderstand.  These are not improperly oriented displays, they are warped, stretched, distorted, broken.  One can barely discern what one is looking at.  I described gaps in the plane's fuselage that shouldn't be there.  I conjecture cockpit view is the view of the pilot strapped in his seat so he should not be seeing under the plane.  And let me emphasize, the wheel is really all I could make out in that display.  Even the display that I believe is the cockpit shows no cockpit features, no controls, no dials.

sthalik, 2017-04-07:
What happens if you completely disable position changes? Set X, Y, Z to "disabled" in the options menu on the "remap" widget.

sthalik, 2017-04-07:
It's sadly the reality of civilian flight sims, that the user can clip through the inside of the aircraft. I'd like to make sure whether it's only position that's broken, before taking any action. I need to know what I'm dealing with.

rayinny, 2017-04-07:
I apologize for the delay in responding.  I appreciate your need for thoroughness and it looks like I'm doing a poor job at conveying what is going on. Perhaps you have to replicate the problem yourself to truly appreciate it.  I emphasize again that this has nothing to do with looking through the fuselage. The display is mostly warped and distorted permanently. You cannot fix it by changing your view.  Let me add this to see if it helps.  If you stop head tracking in OpenTrack, or even close OpenTrack entirely and go back to changing view direction natively in FlightGear, the display remains distorted.  You are just changing views on a distorted display.  Since this problem does not occur in FlightGear without the use of OpenTrack, FlightGear's native view direction system should fix any mere view orientation problem caused by OpenTrack.  It does not.  The only way it seems to fix the distortion is to restart FlightGear.  The distortion does not start until you start tracking in OpenTrack and remains after OpenTrack is closed and until you restart FlightGear.  I hope that helped but please let me know any other inquiries you have.

sthalik, 2017-04-07:
Here's a fresh build with several fixes. It has a .pdf extension but once you change it to 7z, it's gonna open as an archive. github doesn't allow for attaching files with arbitrary extensions so I had to change it.

There was one potentially significant issue relating to the datagrams sent. It didn't occur for me but who knows. I fixed several inverted DOF and added a "help" button that opens the `doc/contrib/FlightGear` directory, so that other users can setup the .xml and command-line switcheroo without guesswork like you had to.

Specifically I tested with `FlightGear 2017.1.2`, the 64-bit version for Windows but it has no issue with other platforms, either.

[opentrack-test-20170407-flightgear.7z.remove-pdf.pdf](https://github.com/opentrack/opentrack/files/906674/opentrack-test-20170407-flightgear.7z.remove-pdf.pdf)

rayinny, 2017-04-07:
Great Sthalik.  Unfortunately I wont be able to test it until tomorrow.  Will let you know the results but thanks for the effort no matter what.

rayinny, 2017-04-11:
My apologies again Sthalik for the delay in responding.  There was a pressing family issue that had to be attended to over several days.

Thanks for the OpenTrack update.  It does cause improvement.  The FlightGear displays are no longer warped but there is even more missing parts of the fuselage.  In fact there is hardly any fuselage now.  Again once tracking in OpenTrack starts, the FlightGear view breaks as described and this persists for almost all the FlightGear views.  It appears the only FlightGear views that are not affected are views for which you can’t change perspective so head tracking wouldn’t apply.

But here is the thing, I NOW BELIEVE THE PROBLEM IS REALLY A FLIGHTGEAR PROBLEM.  After closing OpenTrack, uninstalling it from FlightGear, and persistently cycling through each view and playing around with changing the perspective in each view, I DID FIND A VIEW WHICH WAS BROKEN IN THE VERY SAME WAY I THOUGHT WAS CAUSED BY OPENTRACK TRACKING, THE SAME WAY I DESCRIBE ABOVE.  The only thing OpenTrack apparently does is cause many more views to break in this way.

As I said in my first post, I wasn’t really qualified to judge in which program the problem was sourced and was asking for help in finding this out.  Well seems like we are arriving at a verdict.  What do you think?

sthalik, 2017-04-11:
> As I said in my first post, I wasn’t really qualified to judge in which program the problem was sourced and was asking for help in finding this out.

Don't we all? Cheers.

Are you using third-party aircraft? Are you able to rotate in the 3D pit with the mouse without opentrack?

I recommend going to fgfs IRC. Guys are fun, and once I got in touch and contributed some build system fixes.

sthalik, 2017-09-07:
We're handling this in #682. This is a Flightgear limitation.

closed 2017-09-07 by sthalik