opentrack/opentrack #591: [Feature] Add headtrack output for GoPro VR Player

r57zone, 2017-04-07:
In the settings need to change this:

1. In the Master / Slave settings -> Slave.
2. URL Decoding -> Absolute.
3. Camera Arientation -> Absolute.

After that you can send the json package (only head orientation) to 7755 UDP Port:

_{"fov": 90, "id": "ked", "pitch": MyPitch, "position": 0, "roll": 0, 
"state": 0, "url": "", "yaw": MyYaw}_

Where "MyPitch" and "MyYaw" are yaw and pitch degrees from OpenTrack are 
converted into radians.

MyPitch = DegToRad ( OpenTrack->Pitch )
MyYaw = DegToRad ( OpenTrack->Yaw )

r57zone, 2017-06-19:
[image]
FOV slider (min - 80, max - 120).

It is enough to copy the plug-in Output - UDP over Network, change the port to 7755, add slider fov and change the type of sending from package (float x, y, z, yaw, pitch, roll) to {"fov": MyFov, "id": "ked", "pitch": MyPitch, "position": 0, "roll": MyRoll, "state": 0, "url": "", "yaw": MyYaw}

sthalik, 2017-06-20:
Is it JSON text or binary data?

r57zone, 2017-06-20:
JSON text

r57zone, 2017-08-21:
[Proxy app](https://github.com/r57zone/VR-tracking-apps) if someone else needs it.

sthalik, 2017-12-02:
Sorry it's taking so long. Nowadays my time and energy are limited...

r57zone, 2017-12-02:
I think can close, if anyone needs it, there is a proxy app.

closed 2017-12-02 by r57zone