opentrack/opentrack #592: [Feature] Add headtrack output for SteamVR

r57zone, 2017-04-07:
In SteamVR there is a [Sample Null Driver](https://github.com/ValveSoftware/openvr/blob/master/samples/driver_sample/driver_sample.cpp), which with the help of the [config](https://pastebin.com/raw/UU3BE5BM) (...\Steam\config\steamapps.vrmanifest) allows you to output a stereo image on the desktop screen.

It would be great add support for OpenTrack to this driver.

I added code to the sample null driver, a set the test values sensors (accelerometer, gyroscope and magnetometer) and also quaterion calculate. If the HMD is changed position in SteamVR apps by default, then it will remain only remains to add read from the output UDP Receiver or new output filter for SteamVR (yaw, pitch and roll convert to x, y, z, and to convert to quaterion). ("...\Steam\steamapps\common\SteamVR\drivers\null\bin\win32\driver_null.dll" and "Steam\steamapps\common\SteamVR\drivers\null\bin\win64\driver_null.dll".)

But I can dont get build libraries for Windows (clang + mingw). Perhaps someone can help build this intermediate library for tests?

Thanks.
[driver_sample.zip](https://github.com/opentrack/opentrack/files/905294/driver_sample.zip)

r57zone, 2017-04-07:
Сan also do an make automatic SteamVR configuration. If when selected "Output SteamVR" then check the registry for the "HKEY_CURRENT_USER \ Software \ Valve \ Steam" branch and read value of "SteamPath". If value not null then replace the config and null driver with your own.

sthalik, 2017-04-07:
It's outside the scope of the project. opentrack only sends and receives pose data.

closed 2017-04-07 by sthalik