opentrack/opentrack #594: Custom Center Pose

SoulMaril, 2017-04-09:
Hey,

When you set a Custom Center Pose it does not have any effect on changing the center pose  instead it displaces the red cross of calibration.

Len62, 2017-04-09:
Since the red cross represents the center, wouldn't it be "custom" if it is offset ?

sthalik, 2017-04-09:
Red cross isn't meant to move since it's emitted by the tracker. Center pose only takes place after all of tracker's processing takes place.

SoulMaril, 2017-04-09:
I just want to offset my view in the game. I was using this feature in WarThunder, IL2 Sturmovik to offset my gunsight  a little bit up on the Y axis. Now it is not working anymore. As far as I withnessed when I enter some offset value in that box, say 2 cm, I found out that the Red dot is moving downwards. Try yourself.

If there is another way to offset the view in game tell me. I will appreciate.

sthalik, 2017-04-10:
Hey,

It doesn't occur in unstable branch anymore. Please see the following build <<https://github.com/opentrack/opentrack/issues/352#issuecomment-292789179>>. You need to change the extension from `pdf` to `7z` in order to extract.

closed 2017-12-21 by sthalik