opentrack/opentrack #596: [Bug] Setup Path

SoulMaril, 2017-04-09:
After installation opentrack setup asks to start the executable right away. If you have changed the setup path samething different than the defaul path then it cant launch and gives an error. "Can't find". Setup path seems to be hardcoded.

Len62, 2017-04-09:
Probably set that way for a reason. You can always d/l the manual install version and unzip it to any folder you like.

SoulMaril, 2017-04-09:
I just report. I am already using in a different path without any problems.

sthalik, 2017-04-09:
No, it's not meant to be like that.

SoulMaril, 2017-04-09:
It is not a big deal. It is working anyway. But everytime I use the installer and update it gives this error at first complimentary launch which will never be needed again. Once and all.

closed 2020-01-27 by sthalik