opentrack/opentrack #598: [Bug] Opentrack tries to create a file in D: drive, error message when it fails

zitron-git, 2017-04-12:
OK this is a bit weird and annoying. My D: drive is a SD Card reader. When no SD Card is plugged in, and I start opentrack, I get a error "no disk". I can click continue and keep using OT just fine, but the error goes away if I plug in a card. I found the work around to this is to re-name the drive letter.

To track down what is going on, I used process explorer. OT tries to write a file to the D: drive every time it's open:

http://imgur.com/a/ZkMHa

sthalik, 2017-04-13:
Hey,

I reproduced your result with `procmon` also. Note that `CreateFile` is a generic API for opening a file or creating it -- opentrack doesn't create its own files outside the documents directory.

Get the build at http://opentrack.misaki.pl -- the filename is `opentrack-test-20170413-wip.7z`. It should get rid of all accesses to compile-time Qt install prefix.

zitron-git, 2017-04-13:
Great thank you!

closed 2017-04-14 by sthalik