opentrack/opentrack #599: Install opentrack on Ubuntu 16.04

db4z, 2017-04-15:
Hi,
I'm doing a project and I'm having problems, I can't install opentrack. 
**The project is:**
Use the head tracking device in the airplane simulator software (x-plane 11) on  Ubuntu 16.04.

The developers of the www.trackhat.org device do not support Linux, but in contact with them, they told me that it works with the opentrack available on github.com.
However I can't install opentrack on my Ubuntu. I searched on Google about .deb package to facilitate this process but I didn't find it.

How can I install through source code? I would like some help to perform this installation.

**Head tracking device:** Trackhat Plus - https://www.trackhat.org/trackhat-plus
**Operating System:** Ubuntu 16.04 LTS 64bits
**Flight Simulator Software:** X-plane 11

Thanks!

sthalik, 2017-04-17:
Hey,

You need to make a build directory using `cmake` then build it using `make(1)` or `ninja`, depending on the generator you chose. It's always necessary to run the install target, and I recommend setting the install prefix to a subdirectory of the build directory.

There are no .deb packages because no one contributed code to generate them.

Building on Linux isn't supported (though I fix errors as they become known).

It's possible to using PS3 Eye with 60 Hz as input, and an X-Plane plugin together on Linux. There's also Wine glue to make Windows games detect opentrack on Linux.

- - -

Building on Linux works for the sake of Valgrind, and because writing portable code automatically improves code quality. But Linux isn't supported in the opentrack project like Windows is, as a general-purpose desktop OS. Furthermore, no one volunteered toward maintaining Linux support.

closed 2017-12-02 by sthalik

wleoncio, 2018-03-13:
Have you tried the instructions from [this page](http://tkoletivrijedi.in.rs/node/219?language=en)? They've worked pretty well on my Ubuntu 16.04.4 machine, although for opentrack 2.3.9 instead of the current release 2.3.10. IIRC correctly, the hardest part was installing OpenCV, because that required updating my cmake by compiling a new version. Can't remember the last time installing something on Linux was so much trouble! :-\

sthalik, 2018-03-14:
@wleoncio you can use OpenCV provided with your distribution. Unless it's really ancient, it should work. I haven't tested the 2.4 branch but any 3.x should work.

You should be able to use even packaged OpenCV that depends on Qt4. It wasn't previously possible but now opentrack only links against modules that don't link with graphics libraries.

Unfortunately I don't do more than compile-test Linux nowadays. But nowadays there's also very few platform-specific changes. Code that runs, runs on all platforms so it *should* work on Linux regardless.

sthalik, 2018-03-14:
Finally, building the `unstable` branch is better at most times. Don't bother with tagged versions. There are some times where `unstable` is buggy but normally it's better. It's pretty quiet and bug-free as of right now.

wleoncio, 2020-07-31:
For the record, here's the link I used as a guide to install OpenCV on Ubuntu 18.04: https://linuxize.com/post/how-to-install-opencv-on-ubuntu-18-04/#install-opencv-from-the-ubuntu-repository