opentrack/opentrack #600: Sticky game data after 360 degree rotation using joystick input

evan-king, 2017-04-15:
opentrack 2.3.1.5 on Windows 10, using joystick input.

Move any axis from smallest angle (-180 degrees raw) to its largest (180 degrees raw) fairly quickly* and the game data for that axis sticks at the smallest mapped value it can represent.  Use of mapping curves has no impact other than limiting that value.  Raw tracker data remains responsive, but tracking must be restarted or recentering triggered to fix.

Choosing between relative and absolute input also makes no difference.

\*The faster, the more likely to reproduce, but anythign under ~250ms is fairly reliable - do it as fast as you can and it'll always occur.

sthalik, 2017-04-16:
This is a bug in Accela, it seems. Can you confirm it doesn't happen with other or no filter?

sthalik, 2017-04-16:
There was an error in the formula. Moreover, the Accela filter works badly with sign changes due to operating by means of a delay.

I pushed a build at http://opentrack.misaki.pl as `opentrack-test-20170417-wip.7z`.

evan-king, 2017-04-17:
For some reason, I didn't even notice the filter dropdown before now.  I'll have a play with that next chance I get, which will probably be tomorrow at the earliest.  It was on Accela.

Seeing the functionality the filters offer, I may be able to make better use of my phone's gyros and ditch the joystick anyway.

sthalik, 2017-04-17:
You shouldn't need a filter at all for gyro/mag/acc filtering, basing on what existing users said.

evan-king, 2017-04-20:
For my particular use-case I did in fact need a filter to smooth out the input (particularly to stabilize it when trying to hold still).  However the next filter down worked just fine.

Maybe you want to keep this open to track your bugfix - from my perspective the issue is resolved.

sthalik, 2017-04-20:
I'll keep it open as you suggest. I can reproduce it with vjoystick by playing with its sliders.

closed 2020-01-27 by sthalik