opentrack/opentrack #601: Filter Accela. Changed the max values?

olegiy, 2017-04-19:
Changed the max values?
[https://yadi.sk/i/cj1nl4kQ3H8Euz](url)

sthalik, 2017-04-19:
I did. The previous max values were too high.

sthalik, 2017-04-19:
The limited range has a nice side effect that users will probably feel less lost in the settings. It's easier to find a good slider combination if the range isn't overly broad.

olegiy, 2017-04-19:
My settings did not fit in the new range. :-(  15-20%

sthalik, 2017-04-19:
Why would you need that high settings?

olegiy, 2017-04-20:
I like very smooth )))

sthalik, 2017-04-20:
At 30 ms it's very laggy as if i took too many drugs mixed with alcohol. But you can use the nonlinearity slider to get it smooth also.

olegiy, 2017-04-20:
I agree!

sthalik, 2017-04-20:
Is that OK for you? If not, how much additional smoothing would you need?

olegiy, 2017-04-21:
+15%. Thanks!

sthalik, 2017-04-30:
#605 makes smoothing work on a different scale. It also seems to perform with less lag and more comfort. It involves heavier changes to Accela. See if it works for you.

sthalik, 2017-04-30:
I'm closing this issue since the 6 * 1DOF Accela behavior is hopefully gone forever.

closed 2017-04-30 by sthalik