opentrack/opentrack #603: No image from webcam Logitech C270 on Windows 10 14393.1066

KonstantinKeshikov, 2017-04-30:
Hello.

I have Logitech C270 webcam. Opentrack used to work fine with it. But yesterday it suddenly stopped recieving image from camera. It shows only black picture. The camera is working, the green led is on.
I checked webcam in other program and it works fine. 
Tryed reinstalling webcam drivers and Opentrack with no luck.

sthalik, 2017-04-30:
Switch the USB port for the camera. It should start working.

KonstantinKeshikov, 2017-04-30:
Switching USB port didn't help.
The cause of the problem was my Avast antivirus. I tried to add Opentrack to it's exceptions, but that didn't help. After i completely uninstalled Avast, Opentrack started working fine.

closed 2017-04-30 by KonstantinKeshikov

sthalik, 2017-04-30:
:+1: I'm glad you were able to solve the problem.