opentrack/opentrack #604: black screen issue

chamol, 2017-04-30:
I am using opentrack for sveral years now with DIY IR Led clip and I've never faced any issues until yestraday.

I did not changed a thing in my configuration (win7, opentrack 2.3.1.5, webcam logitech c525) or settings. 

When I click on start, the light on the camera is on, but the screen in opentrack stays black. In the settings of pointtracker 1.1, FPS @1fps.
What I checked so far:
- webcam is working in VLC, skype and leds working in CL-eye. 
- tried to work with freetrack and facenoir with same issue
- tried to roll back opentrack with previous versions -> same issue
- tried to install other codec -> same issue
- tried other setting in webcam setting -> same issue
- tried todesinstall/reinstall webcam drivers -> same issue
- checked if any other program was using the webcam -> I've seen nothing. 

Any more ideas ? I suspect a problem in pointtracker but...don't know how to tacle it...

Len62, 2017-04-30:
Someone else on another forum was having the same issue and tracked it down to a conflict w/AVG Antivirus. The person uninstalled it and all was fine.

chamol, 2017-04-30:
thx for the tip.
De-activated avast but did not work.

Sgt-Karl, 2017-05-01:
chamol, I had the same issue(Camera worked in Skype, OpenBroadcast Software and other Camera Apps). So, I disabled the Avast shields but didn't work then I uninstalled the whole Avast software and it worked. OpenTrack is picking up my camera now.

chamol, 2017-05-03:
thanks for your help.
I uninstalled Avast but i did not worked...

chamol, 2017-05-05:
Ok I found something that helped.
1) Uninstall webcam drivers (logitech c525)
2) suppress logitech entry in registery (regedit -> search logitech)
3) reboot
4) plug c525
5) detection in opentrack(and facetranoir and freetrac) back on 

Fixed the isssue.

Len62, 2017-05-05:
Great!

sthalik, 2017-05-06:
@chamol it might be equivalent to changing the driver to "UVC Video" in Device Manager without registry changes. I'm happy that it works for you though. I still think it's a codec issue since my c525 works.

Azza276, 2017-05-07:
I got this working with Avast Installed. This is not a fix for OT, but for record, and to help others, maybe even to put in your troubleshooting/Help guides...

No driver screwing around and no regedits.

You need to add OT to the exclusions list in Avast, not the General/exclusions part in the settings menu, but Ii's actually in the "Behavior Shield" Component. 
1. Click on settings
2. Click on Components
3. On Behavior Shield, click on "Customise"
4. Click on Exclusions
5. Browse for the opentrack folder.
6. Click Add.

For this I added the whole folder for testing just to cover all the dll's etc.

sthalik, 2017-05-07:
@Azza276 thank you for the instructions. Unfortunately no one reads the FAQ.

sthalik, 2017-05-07:
I think we're done here, no one seems to have a problem anymore.

closed 2017-05-07 by sthalik

Azza276, 2017-05-07:
I Read the FAQ's... Sometimes :) No worries.

eldarbr, 2020-04-01:
Have same issue, but with cam paanasonic hx-a1. Never had Avast installed. Camera works in all other programs, including freetrack. I added opentrack folder to windows defender exclusions, but it didn't helped, starting program as admin didn't too. windows 10

clonezzz, 2020-05-22:
> 
> 
> Have same issue, but with cam panasonic hx-a1. Never had Avast installed. Camera works in all other programs, including freetrack. I added opentrack folder to windows defender exclusions, but it didn't helped, starting program as admin didn't too. windows 10

1 install manycam
2 set source as hx-a1 cam on manycam
3 Set source on opentrack as manycam

eldarbr, 2020-05-22:
The issue was resolved with webcamoid (https://github.com/webcamoid/webcamoid). Thanks to everyone