opentrack/opentrack #605: opentrack 2.3.1.6 release process

sthalik, 2017-04-30:
Hey,

I'm soon going to release opentrack `2.3.1.6`. I'd like the dedicated users, and perhaps others, to submit feedback. As you know this hobby project has no QA team.

@mrsanchos has officially joined the development team. Welcome! :cat2: 

Some of the changes you might like:

- **fix Vive rotation order and enable this tracker.** fixes provided by @subnet- and @antcodd, thanks!
- **accela: now filters 3 DOF for rotation and position rather than each separately**; please report new issues!
- accela: reduce sudden jumps on minimal movement (reported by and discussed with @mrsanchos)
- remove camera angle option that wasn't doing anything
- accela: reduce acceptable value range; this makes it easier to configure
- reduce CPU usage while tracking, spent in program's core functionality
- fixed the path to Intel RealSense camera driver installer, by @ph0b 
- flightgear: some output DOFs were inverted
- allow for roll calibration; it mostly works from my testing
- some operations in the PT and Aruco trackers are now faster thanks to the Intel IPP library
- fix PT point size display; make calibrated center a yellow cross

Grab builds at <<http://opentrack.misaki.pl>>. The current build filename is `opentrack-test-20170430-2.3.1.6-pre0.7z`.

sthalik, 2017-04-30:
Paging @MathijsG @Len62 @mrsanchos @olegiy @petsild. Cheers!

mrsanchos, 2017-04-30:
@sthalik thx Stas! :)

sthalik, 2017-04-30:
That was fast :)

Len62, 2017-04-30:
A quick test looked good. Will test some more tomorrow.

ph0b, 2017-05-02:
I've encountered an issue while testing so I had to do a small fix, sorry! https://github.com/opentrack/opentrack/commit/ff3f20bf50ccae00937732673bbb12f1b6be7afe

@sthalik did you change anything regarding unicode support in plugin names or the build itself?
I had no such issue when building with msvc2015 or with previous releases, but in your test build I get:

[image]

sthalik, 2017-05-02:
@ph0b previously it was random whether utf-8 literals worked. Now they should always work, provided they're in form `u8"foo"`. The type is `const char*` of course.

sthalik, 2017-05-02:
The regression is 36540c773f6833696bae3364c39217847fc5f96b.

sthalik, 2017-05-03:
@ph0b I was actually confused by a comment in `kalman.cpp` to that extent. It appears we don't actually have issues of this sort. Carry on please, no changes to string literals are necessary at all.

sthalik, 2017-05-03:
I updated the test build in the first post. It's `opentrack-test-20170503-2.3.1.6-pre1.7z`. The changes are:

- fix realsense driver installer path -- by @ph0b 
- undo the Unicode mess -- me

With regard to the Unicode change, there shouldn't be any user-facing changes.

ph0b, 2017-05-03:
@sthalik perfect! your build is working well for me now, thanks.

sthalik, 2017-05-03:
@ph0b please see if my cmake changes broke anything, thanks. It now unconditionally generates and installs `.pdb`s. The rationale is that users can't debug unless they have the files, and by the time a build's released, the files are long gone.

If this should be conditionalized let me know.

Len62, 2017-05-03:
Newest build is working great! No issues from my end.

MathijsG, 2017-05-10:
Will try soon. Quite busy with some other work unfortunately. One minor question, are we diverting from semantic versioning already? :-P hence the extra digit. Shouldn't this be 2.3.2?

sthalik, 2017-05-14:
@MathijsG I will drop the penultimate digit.

sdysj, 2017-05-15:
Is it just me can't get it work on DCS World latest version? 
It worked before. The options page shows up TrackIR correctly but just don't work.
BTW using the dll from https://github.com/MaximeMorel/DCS-headtracker works.

sthalik, 2017-05-15:
There were no changes to that module. Same dll, same logic sharing with the dll, same game list.

sthalik, 2017-05-15:
Do you get a game name on opentrack's titlebar?

sdysj, 2017-05-15:
@sthalik No, the titlebar just show profile name.

Len62, 2017-05-15:
Hmm, just loaded up DCS World here and OT works fine. Are you running OT as administrator? You could also try doing a repair of DCS World. When running OT the titlebar shows the profile, and then Black Shark.

sthalik, 2017-05-15:
Disable "stop tracking on program exit".

See if the registry path exists -- `HKCU\NATURALPOINT\NPClient Location`. It must point to where `NPClient64.dll` is stored.

Some people also said it suddenly starts working after rebooting.

sdysj, 2017-05-15:
Solved. Reset the profile got it work.

PH-LCS, 2017-06-28:
Hi,

Opentrack 2.3.1.6 is not working for Prepar3D v4.
See head moving in Opentrack but no tracking in Prepar3D v4.
All programs running as administrator.

closed 2018-01-04 by sthalik