opentrack/opentrack #606: OT and antivirus issues

Len62, 2017-05-01:
Not a problem for me but a couple of OT users on other forums have run into issues with OT and AVG/Avast. Only way they could get OT to work properly was to uninstall the AVG/Avast programs. Just wanted to pass this along.

sthalik, 2017-05-01:
The antivirus software blocks camera use by default apparently. Good from their point of view but nothing opentrack can do about it.

petsild, 2017-05-01:
If anyone has a problem just add OT to the list of exceptions.
[image]

Len62, 2017-05-01:
@sthalik The users reported no other issues using their cameras. 
@petsild The users reported that just excluding OT did not help.

Never had this problem myself, of course I don't use AVG or Avast.

closed 2017-05-01 by Len62

sthalik, 2017-05-02:
@Len62 there's nothing that can be done from programming perspective.