opentrack/opentrack #607: Tracking seems to time out after about five minutes on Galaxy S7 Edge

HeavyHandedHugs, 2017-05-02:


sthalik, 2017-05-02:
It's likely some battery optimization. We try to keep a wakelock over the cpu and wifi but it might not be enough. Add the freepie .apk to the list of exceptions.

HeavyHandedHugs, 2017-05-03:
Okay, I added it to the sleep exception list but I tested it for about an hour today and it's still doing the same thing.

sthalik, 2017-05-03:
If there's a "sleep exception list" that itself implies that the vendor messes with wakelocks.

But I can think of a workaround:

- go to development options
- enable "don't sleep" -> "while usb debugging"
- enable remote ADB debugging
- on the PC, run `adb connect phone-wifi-address`
- on the PC, run `adb shell` and confirm on the phone. you should get a shell.

HeavyHandedHugs, 2017-05-03:
- I can't find "don't sleep" -> "while usb debugging" in the development options. I turned on USB debugging, though.

- I can't find anything that has ADB in it except:

    "Verify apps via USB" "Check installed apps from ADB/ADT for harmful behaviour" which I unchecked.

- I'm assuming you want me to run "adb connect phone-wifi-address" in command prompt? When I do, it returns "'adb' is not recognized as an internal or external command, operable program or batch file." I also tried it in "Run" but it said windows couldn't find ADB.

sthalik, 2017-05-04:
@HeavyHandedHugs I was looking for a sure way to prevent your phone from sleeping. Technically it should be enough as it is.

Try turning off battery optimization completely for the duration of using the software.

closed 2017-12-02 by sthalik