opentrack/opentrack #608: Hot Keys to enable/disable axis

Len62, 2017-05-03:
Was assisting another OT user in getting OT set-up. He noticed that OT does not have hot-keys available to enable/disable the individual axis. Not sure when this would come in handy, but I figured I'd post it here.

sthalik, 2017-05-03:
What's the use-case here? Does similar software have a use-case for this? Does it need multiple keystrokes for different toggles?

Len62, 2017-05-09:
I never heard back from the user who inquired about this. In his original post he mentioned something about centering in freetrack and I don't think that issue exists in OT. Given the options available in OT, centering is handled very well. I really didn't see a need for this, but you never know.

sthalik, 2017-05-14:
There were use-cases to prevent strain over longer simming sessions. But it's pretty obscure as it is.

I'm closing this until hearing a fair rationale, and too bad that user wasted your time.

closed 2017-05-14 by sthalik

Input-BDF, 2024-06-19:
Came over this issue as I am actually in a situation, where it whould be usefull to lock certain axes or switch between a set of active and inactive assignments.

Usecase would be in my case related to Armed Assault 3 as example.
Opentrack comes extremly handy as pilot (using all 6 axes) and I really enjoy it. But when you are often switching between pilot and groundforces position (as I do as SL) OT using all 6 axes makes the infantry-part nearly impossible. As infantry only Yaw, Pitch is usefull. Sometimes I would rather stick back to only use yaw to look arround quickly. Same comes into account with groundvehicles, where the full portfolio of axes are more obstructive. Think in other games (out of the aviation scope) it's the same.

If implementation of hotkeys for enable/disable axes, is to much work and interferes with some of your concepts, maybe a soft-solution is to introduce a hotkey to toggle between two profiles by stop tracking, switch profile, (re)start tracking. Out of this maybe come some other yet unknown usecases as (in my example) wired different headmovement min/max abilities/centers in different Arma planes.

Nonetheless, this is probably more of a "quality of life feature" request. Or something which addes unique features to OT other users even yet don't know they could need ;)

GO63-samara, 2025-10-02:
Maybe try like on the screen or something like that:

[image]