opentrack/opentrack #610: Tracking not working using profile from 2.2

Anghostdy, 2017-05-07:
I try to use my opentrack profile from 2.2 in version 2.3.6.1. Settings are used from 2.2, but nothing works. I checked the settings manually and it seems fine.

What can be the issue here?

Len62, 2017-05-07:
There have been enough changes between 2.2 and 2.3.6.1 that you should just create a new profile. Various aspects of OT have been improved and it can create issues when using old profiles.

sthalik, 2017-05-07:
Starting with the 2.3 series, setting incompatibilities get taken care of. That doesn't apply to 2.2 series though.

What @Len62 said is all true.

MathijsG, 2017-06-07:
Is there maybe a way to warn for this? Or somehow filter out the profiles that have a version too low in their file? I sometimes test/play many different games with headtracking, resulting in sometimes having a profile from a few years old. It's not a big problem, but maybe if we are starting to use more strict versioning and we exactly know when an API/function breaks and when it doesn't, we could be filtering out the incompatible profiles. Or maybe even converting them.

Again, very minor :-), just something which would make it more foolproof if we had unlimited devtime.

closed 2017-12-02 by sthalik