opentrack/opentrack #611: Black-Screen Webcam on Aruco-Tracker

MobyX521, 2017-05-07:
A few weeks back, I tested out the Aruco Tracker to track my head during flight simulations. I was very pleased, but put the project off to the side for a few weeks so I could have a better mount for the Aruco marker on my head. 

Well, here I am today, ready to use the Aruco Tracker, but my webcam will not work. I am using a logitech B910 with standard windows drivers. (plug-n'-play) I have also tried using the logitech drivers. When using the webcam in outside programs or in my web-browser, it works fine, but with the Aruco Tracker on opentrack and with the FacetrackNoIR program, I get no picture, and only see "Hz: 1" displayed in the top-right corner. The light for my webcam turns on and everything, I just get no picture.

I am currently using version 2.3.1.6 but have also tried using version 2.3.1.5.

Any help would be appreciated.

sthalik, 2017-05-07:
Use "ManyCam" to pass video from your real webcam. This solves the problem due to codec differences.

The bug #457 tracks progress toward the real solution, i.e. performing camera input the same way Skype and other software do. But it's a lot of work any we don't have any (other) volunteers.

sthalik, 2017-05-07:
It also helps sometimes to change the USB port for the camera. This forces Windows to reinitialize the camera's driver.

MobyX521, 2017-05-07:
Please specify what you mean. I have installed ManyCam but it seems like it's just an ordinary webcam tool. How will this fix my issue?

MobyX521, 2017-05-07:
Nevermind. Thanks for the fix! I really appreciate what you do!

closed 2017-05-07 by sthalik