opentrack/opentrack #615: Issue building opentrack on linux

winiciuscota, 2017-05-10:
I'm a newbie to the build process, this is what I did:

git clone https://github.com/opentrack/opentrack
cd opentrack
mkdir build
cd build
cmake ..
make

Got the following error

cc: error: unrecognized command line option ‘-fno-lto-fvisibility=hidden’; did you mean ‘-fvisibility=hidden’?
make[2]: *** [CMakeFiles/opentrack-version.dir/build.make:63: CMakeFiles/opentrack-version.dir/version.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:68: CMakeFiles/opentrack-version.dir/all] Error 2
make: *** [Makefile:128: all] Error 2

You said the build process does supports linux. Am I missing something?

miniskipper, 2017-05-11:
Came across the same error. I build with ninja, so I edited build.ninja and replaced 
`-fno-lto-fvisibility=hidden`  with  `-fno-lto -fvisibility=hidden`

In your case its somewhere in a .make file

miniskipper, 2017-05-11:
In  opentrack/x-plane-plugin/CMakeLists.txt change 
`set_property(${begin} LINK_FLAGS "-undefined_warning")`
to
`set_property(${begin} LINK_FLAGS " -undefined_warning ")`
(spaces)

In  opentrack/cmake/opentrack-version.cmake change
`set_property(TARGET opentrack-version APPEND_STRING PROPERTY COMPILE_FLAGS "-fno-lto")`
to
`set_property(TARGET opentrack-version APPEND_STRING PROPERTY COMPILE_FLAGS " -fno-lto ")`
(again, spaces)

sthalik, 2017-05-14:
@miniskipper thank you, I'll make a macro for it given how brittle it is.

:+1: :cat2:

closed 2017-05-14 by sthalik