opentrack/opentrack #616: Logitech C615

raviv, 2017-05-10:
Hi,

I'm trying to use opentrack point tracker with a Logitech C615 webcam and unlike what i see in tutorials, the video input looks normal (as in not black) and it picks up many points in the picture, not just me LEDs.
What am I doing wrong?

Thanks,
Raviv

Len62, 2017-05-11:
Open the Point Tracker settings and adjust the "Point Extraction" settings so only the LEDs get picked-up. Are you using IR or regular LEDs?

MathijsG, 2017-05-14:
Did you remove the IR filter and blocked light by using some kind of filter? For example a piece of film or a floppy diskette piece? By default a regular webcam picks up a lot of daylight and blocks the IR.

closed 2018-01-04 by sthalik