opentrack/opentrack #620: Acella

ljgdasfhk, 2017-05-14:
Akella does not work
The image jerks even at maximum settings

sthalik, 2017-05-14:
Increase nonlinearity, remove smoothing. Smoothing doesn't remove noise.

Are your model dimensions accurate?

ljgdasfhk, 2017-05-14:
The dimensions of the model have not changed since the previous version 2.3.1.4
Something does not work as before
[https://www.youtube.com/watch?v=Zk9Czvx5fwk](url)

sthalik, 2017-05-15:
in `filter-accela/accela-settings.hpp`, change:

`rot_sensitivity(b, "rotation-sensitivity", slider_value(1.5, .1, 2)),`

to

`rot_sensitivity(b, "rotation-sensitivity", slider_value(1.5, .1, 4)),`

The last value is rotation sensitivity. The same you can do for `pos_sensitivity`.

ljgdasfhk, 2017-05-15:
In my opinion, the filter Akella does not work.
Watch my video
https://www.youtube.com/watch?v=9yPdR5GI2vE
No filter or selected achele - no difference.

sthalik, 2017-05-16:
I'll do some tests on Linux. It might be related to time delta but this is a long shot.

sthalik, 2017-05-16:
Wow, it's really like that!

See line 157 here: <<https://github.com/opentrack/opentrack/blob/632cd5bf7778cb9062633f8d27ecd6aadcaa5d28/filter-accela/ftnoir_filter_accela.cpp#L157>>.

Replace `< 0` with `>= 0` and it should start working. Tell me if it helps.

ljgdasfhk, 2017-05-16:
Now everything works fine.

closed 2017-05-16 by ljgdasfhk

sthalik, 2017-05-17:
@ljgdasfhk correct fix is to remove these 4 lines. The if/else and fmax/fmin lines.