opentrack/opentrack #621: Combination of Vjoy

petsild, 2017-05-14:
Hi,
when 2.3.1.5-6 setting  a combination of input – PT and output - Vjoy, occurs accidental falls application. The same problem have friends with different running W7,10 and other hw, thank you. Pet

sthalik, 2017-05-15:
What exactly do you do with PT and vjoystick?

petsild, 2017-05-15:
I use vJoy device 1 to control the virtual axis X,Y,Z for the game PlanetSide 2. It works but sometimes it randomly falls off, someone 2x per day others more often.
But running the game the error does not affect, thanks for any resolve.  Pet

sthalik, 2017-05-15:
Please install Visual Studio 2017 Preview or even 2015 Update 3. We'll have to see where it crashes.

cheers

petsild, 2017-05-15:
What all do I install a summary of the dozens of components?
[image]

sthalik, 2017-05-16:
Desktopowych aplikacji pomocy C++

_Nie_ instaluj uniwersalnej platformy. Potrzeba tylko środowiska i obsługi C++.

:laughing:

petsild, 2017-05-16:
I installed the Community 2017, what do I do next?

sthalik, 2017-05-16:
Start the software. Wait for it to crash. Now start Visual Studio without closing the "program has crashed" window.

- connect to crashed process <<https://msdn.microsoft.com/en-us/library/3s68z0b3.aspx?f=255&MSPPError=-2147217396>>
- press the green button on the toolbar that looks like "pause" on a music player, next to the red "stop" button
- select Debug -> Windows -> Call stack from the menu
- screenshot the call stack window

It might be missing the .pdb files, but don't worry for now. We'll go through loading them if that info's not enough.

closed 2017-09-07 by sthalik