opentrack/opentrack #624: Options other than Freepie UDP Receiver and 'UDP over network' are absent

shahratin, 2017-05-20:
I'm on Fedora 22. I did:
`git clone https://github.com/opentrack/opentrack --recurse-submodules
mkdir build
cd build
cmake ..
make install
cd build/install/bin
./opentrack`

sthalik, 2017-05-21:
Clone this repository with submodules: <<https://github.com/opentrack/opentrack-depends>>.

Use `cmake-gui` or `ccmake` to configure dependency paths. In particular, the `SDK_` prefix is for components such as LibOVR, etc. There are some ungrouped dependencies though.

I heavily recommend you build this fork of opencv rather than link against system version. It's likely that the system-provided version links against Qt4 and it crashes at runtime since we link against Qt5 also.

shahratin, 2017-05-27:
Thanks, got it working now.
The photo module of opencv was not compiling, so i did `cmake -D BUILD_opencv_photo=ON`
Is that module necessary?
Also wondering why it didn't appear in the ccmake.

closed 2017-05-28 by shahratin