opentrack/opentrack #625: False movement interpretation

feanor91, 2017-05-24:
Opentrack version 2.3.1.6
Protocol : trackpoint
Filter : Accelera (tried others too)
Hardware : Dealanclip and PS3 Eye you camera

Every thing works well, except that when I pitch/yaw, I have ghost movement on X, Y or Z axis. If I zomm on Z, this ghost movement on X and/or Y are amplified.

If I define very large deazones on axes, I can trap this default, but it makes X and Y wanted movement almost unusabe of course. I'm talking settinf deazone around 1/3 oth the course.

The systeme behave like when I turn the head, a sliding movement is interpreted.

This behaviour is all but unusable into DCS and clickable cockpit as I can't focus mouse on the point I want to click. 

With the new "Togle view position", I think it will be a turn around, but it would be fine to have a better way than this or deadzone to track this default.

Len62, 2017-05-24:
Have you tried adjusting the accela settings? 
[image]

feanor91, 2017-05-24:
Yes, I looked everywhere and tried a lot of things. Id did something like you show here.

One point, a big lack of documentation about what parameter do what as in Opentrack and filters, the wiki, is not enough to understand clearly what parameter is used for. In particular I think of this :
[image]
and that:
[image]

I'm not native english, and even if I understand the meaning of the text I have no idea of the concepts involved either for the hardware part or the software part. 

As for Accelera filter by the way. To be short parameters part are very dark for me, So, you hope set by default, all will work, and it does, usually, but in some particular case, is not, and it is where you have to dig into parameters to try things, but not understanding what you are doing is a little frustrating :)

Anyway, I read some more sooner, and I saw facetracknoir seem better for DCS, I will have a try, else, I don't know;

feanor91, 2017-05-24:
Perhaps one more information I forgot : I'm not close to camera when I play due to mu desk configuration, I think I have at least one meter between camera and clip, perhaps it means something.

sthalik, 2017-05-24:
For rotation and position interconnect, perform calibration in PT tracker options. It's now really easy and you don't need to hurry while doing it. Just follow the simple instructions. You'll get a feedback message after you're done.

Understanding relative translation isn't easy unless you imagine it as a geometrical motion in 3D space. Let me try though -- with it on, turning around to the aircraft seat and moving backward on Z will move you toward the front of the cockpit. This is because rotation influences position changes. With it off, you'll move toward the seat you're looking at, since you physically moved backward. It's a matter of preference which mode you use. You aren't losing anything critical by skipping these settings. Also relative translation can be confusing unless you understand how it works.

You aren't losing anything by skipping the camera tab also. I recommend not enabling stuff you don't understand cause it's going to make stuff behave unpredictably.

Finally, the documentation I personally wrote is too dry and too hard to understand. The [quick start guide](https://github.com/opentrack/opentrack/wiki/Quick-Start-Guide-(WIP)) is entirely written by community members so it's of higher quality.

If one of you could explain relative translation the right way so anyone can understand, please put it on the wiki and send me the link. I'll put the link in the options window.

sthalik, 2017-05-24:
For the calibration process, see <<https://github.com/opentrack/opentrack/wiki/Quick-Start-Guide-(WIP)#video-of-the-calibration-process>>.

feanor91, 2017-05-24:
Hum...Seems I miised this calibration phase. Thanks to point it out, I did it now, and will try in DCS asap.

feanor91, 2017-05-25:
Hi, it was that, once calibrated, all works fine. Thnqks a lot

closed 2017-05-25 by feanor91