opentrack/opentrack #627: Opentrack 2.3.1.6 and P3Dv4

ArnieWe, 2017-06-05:
Hi,

I switched from FaceTrackNoIR to Opentrack 2.3.1.6 to see if it works with the newly released Prepar3d v4.
Everything went fine the first time I ran Opentrack in P3D, so I happily decided to setup profiles for my other sims as well (DCS, rfactor2). 
The sad thing is that everytime I start Prepar3D now it somehow does not respond to headtracking anymore, despite the Octopus moving as it should. There are no error messages either.

Here is what I tried so far:
- reinstalled all three simconnect libraries that shipped with the sim
- tried Opentrack in P3D v3, no problem there
- successfull check with SimConnectTest.exe (found it on google)
- reinstalled Opentrack 2.3.1.6
- uninstalled P3D v3 to see if there was a conflict between the two sims

Since none of the above helped and FTNoIR is also not working I'm wondering if headtracking is supposed to work with the 64bit-Version of P3D at all - even if it did once in my case...

PS: It might be of interest that there are four version of flightsimulator.simconnect in my winSxS folder:
10.0.62613.0
10.0.60905.0
10.0.61242.0
10.0.61259.0
and also one esp.simconnect

PH-LCS, 2017-06-05:
I have the same problem and have no solution so far.

SoulMaril, 2017-06-05:
Not a specific solution for your problem. Just for your information.

I am able to run Opentrack with Prepar3D v4 without any issues. For a general rule of thumb make sure you are running both opentrack and P3D in adminitrator mode.

blamarche, 2017-06-05:
I am also having this issue. Worth noting (and others should add too, if you can): Win10 home

PH-LCS, 2017-06-06:
@SoulMaril 
Please note that both programs are running with administrator rights. And in P3Dv3 it works flewesly.....

SoulMaril, 2017-06-06:
Ok. I found this. I am using EZCA ezdok camera. When I quit ezca opentrack wont work. So yes it is not working for me either. Seems like a simconnect compatibility issue. You can find all the legacy simconnect versions at "..\Prepar3d v4 root\redist\interface".

sthalik, 2017-06-06:
At http://opentrack.misaki.pl you have a test build with simconnect debugging. Start it under "Command Prompt" by dragging opentrack.exe onto the command prompt's black window. Text will arrive. See if there are any errors from simconnect.

bjensie, 2017-06-07:
Hi folks!
I found the following thread on the Prepar3D forum. Maybe that will help out.
http://www.prepar3d.com/forum/viewtopic.php?f=6310&t=124776

@sthalik BTW, would it be possible for you to release both a 32bit and a 64bit build?

sthalik, 2017-06-07:
@bjensie I'll provide a wrapper executable. We already do this sort of thing.

**Please** send me the **P3D SDK** installation directory content after its installation as I don't have the title! Getting the SDK itself won't help as it requires the game, previous SDK versions etc. Installing FSX SDK was always a pain.

sthalik, 2017-06-07:
Update: I only need `SimConnect.dll` from `C:\Windows\Winsxs`. There are multiple versions and I need collect them all, like Pokemon.

sthalik, 2017-06-07:
Now I'm confused. Is `SimConnect.lib` from P3D SDK a static library, not a convenience `.dll` import library? Look at that:

```
Dump of file SimConnect.lib
File Type: LIBRARY
  Summary
           8 .CRT$XCU
          78 .bss
          20 .data
       8DFB7 .debug$S
       37E20 .debug$T
         371 .drectve
          28 .idata$2
          28 .idata$3
          10 .idata$4
          10 .idata$5
          18 .idata$6
         EDC .pdata
         69A .rdata
          26 .text$di
       11A23 .text$mn
         138 .text$x
           E .text$yd
        1980 .xdata
```

The crapola has `.text` and `.bss` sections. Convenience import libraries don't have that. I'll look into it tomorrow or something.

SoulMaril, 2017-06-07:
There is a managed dll library for C# and VB aplications and another for C++ library as SimConnect.h  and SimConnect.lib

More information can be found here: http://www.prepar3d.com/SDKv4/LearningCenter.php

SoulMaril, 2017-06-07:
Managed dll will work no matter if it is 32bit or 64bit but for C++ apps it must be 64bit as far as I understand.

sthalik, 2017-06-07:
@SoulMaril SDK v4 only has 64-bit managed binaries in `lib/SimConnect/managed`. But I'll deal with it through a wrapper executable.

There's no reason to release 64-bit Windows opentrack. It's also less convenient for users. We might even be unable to support some protocols/trackers with 32-bit only SDKs.

SoulMaril, 2017-06-07:
Soruce: http://www.prepar3d.com/SDKv4/sdk/simconnect_api/c_simconnect_projects.html

I was refering to this 

> Ensure that the Solution Platforms, in Visual Studio 2015 or later, have been set to 64-bit. While SimConnect interfaces are backwards compatible with legacy 32-bit SimConnects, all new development should be done as a x64 solution.

**All Projects**
Ensure that the Solution Platforms, in Visual Studio 2015 or later, have been set to 64-bit. While SimConnect interfaces are backwards compatible with legacy 32-bit SimConnects, all new development should be done as a x64 solution.
Ensure the SimConnect.ini file is placed in your %USERPROFILE%\Documents\Prepar3D v4 Files folder. A default version of this files can be found in the SDK\config\SimConnect directory.You do not usually have to make any changes to this file. The SimConnect.ini file provided ensures that the SimConnect debug window opens to display communication information between the server and the clients.
If the client is to work remotely, write a SimConnnect.cfg file for it.
By default, SimConnect is disabled. To enable it, you must copy the necessary configuration files to the correct locations. Default versions of these files can be found in the SDK\config\SimConnect directory. Ensure the EXE.xml, DLL.xml files and SimConnect.xml files are placed in either the %APPDATA%\Lockheed Martin\Prepar3D v4 or %PROGRAMDATA%\Lockheed Martin\Prepar3D v4 folder, and edit these files according to the instructions that follow.
Run Prepar3D to test your SimConnect client.

**Using Legacy SimConnect Versions**
Prepar3D

FSX and ESP

Some third party add-on software requires an older version of SimConnect that was built with either FSX or ESP. When installing Prepar3D through the Prepar3D Bundle, the older SimConnect redistributables are installed automatically. To manually install or reinstall the destributables, go to the redist\Interface folder of the installed product. There are multiple different legacy versions of SimConnect available:

ESPv1
FSX-RTM (Flight Simulator X)
FSX-RTM-JPN (Flight Simulator X, Japanese)
FSX-SP1 (Flight Simulator X: Service Pack 1)
FSX-SP1-RUS (Flight Simulator X: Service Pack 1, Russian)
FSX-SP2-XPACK (Flight Simulator X: Service Pack 2 with Expansion)
Each of the legacy versions of SimConnect can be installed by following the instructions in their root folders. For most, you can install them using their respective .msi files.

After installing legacy SimConnect client libraries, they will be located in the WinSxS (side-by-side) folder. This means that software add-on components written for older versions of SimConnect will still run as newer versions are released.

ArnieWe, 2017-06-07:
> At http://opentrack.misaki.pl you have a test build with simconnect debugging. Start it under "Command Prompt" by dragging opentrack.exe onto the command prompt's black window. Text will arrive. See if there are any errors from simconnect.

i get this error message: simconnect: can't open handle: -2147467259

sthalik, 2017-06-07:
@ArnieWe thanks. Apparently the older simconnect SDK won't work.

Does EZCA come with a separate binary for P3D or is it same as FSX?

ArnieWe, 2017-06-09:
I don't have EZCA. Perhaps SoulMaril can help with this one?

bjensie, 2017-06-09:
Hi again!
I just want to hint that the thread I pointed to above has been updated. In that thread there's also
a link to another thread, concerning executables talking to Simconnect. Here's the link:
http://www.prepar3d.com/forum/viewtopic.php?f=6312&t=125109

I hope this help in some way.

PH-LCS, 2017-06-09:
Allready read it but still no solution ….;-(

 

http://www.prepar3d.com/forum/viewtopic.php?f=6312 <http://www.prepar3d.com/forum/viewtopic.php?f=6312&t=125109> &t=125109

SoulMaril, 2017-06-09:
@ArnieWe @sthalik I am not very good with simconnect. I generally use fsuipc managed dll by Paul Henty. I tried and tested to connect P3Dv4 with his managed dll using C#. Although I build my solution as a x86 application I am able to connect to 64bit v4.

So I can say using a managed dll will connect to P3D v4 no matter if it is 64bit or 32bit so this explains why Ezdok is working with Opentrack. They must be using a managed simconnect.dll

PH-LCS, 2017-06-13:
Hi,

 

Here are the files.

 

Gr.

PH-LCS, 2017-06-19:
Any news on the latest status?

sthalik, 2017-06-19:
No. Don't expect it on short notice. It's actually detrimental if you ask at all.

ArnieWe, 2017-06-19:
I finally got EZCA 2 and installed it into P3Dv3. After copying some files over to v4 (see instructions on the avsim forum) it does the job in combination with opentrack. It even worked with all three simconnect protocols at first launch but no more after opentrack was restarted. Just like i described in my first post... very odd.
Now i'm using the freetrack2.0 protocoll which tells EZCA i'm using TrackIR - this seems to be a stable workaround.

ChristianP3DV4, 2017-06-22:
Hi all!

I would like to inform you briefly that I had similar problems yesterday, to connect my HeadTracker FaceTrackNoIR with the flight simulator P3DV4.

But now the developer of FT, Wim, has created a solution. Wim has programmed a new protocol and now everything works again as before.

The Tracker FT I've been using for many years, first for the FS2004, later for the FSX and last for the P3DV1/2/3/4. But with the release of version 4 (64-bit simulator) FT no longer worked. This was very very annoying. So I tried to move Wim to update the corresponding SimConnect protocol. And he did it, for which I am very glad and grateful. Because now I can operate as usual my simulator and move freely in the cockpit. My thanks to Wim - Developer of FaceTrackNoIR!

Best regards
Christian

[image]

ronh991, 2017-06-28:
I am going to suggest that the scserver manifest files are not correct.  The ver numbers don't agree with the name example scserver is 10.0.61259.0 while scserver - sp2 is 10.0.60905.0 and sc-server accl is 10.0.61242.0

my finding are:

RTM "10.0.60905.0"
SP1 "10.0.61242.0"
SP2ACCEl "10.0.61259.0"

so if you select incorrectly you may think you have accl setup but are really using SP1 and SP1 does not work in P3D v4 and if you select sp2 you are using rtm and RTM does not work in P3Dv4.  So try just the default scserver.

I maybe wrong, but it looks strange to me.

Also the MS VC80 should be 8.0.50727.6195.

Also try running as administrator.

sthalik, 2017-06-28:
Hey,

I won't be changing existing manifest entries due to the fear of regressions. Also, it works for SP2, Acceleration, and Steam.

Can you send me the facetracknoir .dll for simconnect? I'd like to extract the manifest contents from it. It's somewhat likely to make P3D support work.

sthalik, 2017-06-28:
Regretfully I haven't released 2.3.8 yet. Things are hard here and adding a binary release is too much than I can handle.

ronh991, 2017-06-28:
I have a manifest for hidmacros (written id delphi/turbo pascal)  It should be the same as I use microsoft rc to compile it.  I will setup a onedrive link.

ronh991, 2017-06-28:
Here is the link:

[simconnect.manifest](https://1drv.ms/u/s!Am7yhLBF1vbkgT2aCDtHrTGLU28c)

sthalik, 2017-06-28:
Is this working for Prepar3D?

ronh991, 2017-06-28:
This resource file works for me in the hidmacros testing I have done.  I also have a C++ program the works in Prepar3D v3 and v4.  But I am linking to the Prepar3D simconnect.lib files.

The hidmacros loads the dll dynamically.

sthalik, 2017-07-06:
TL;DR Try the SP2 or Acceleration versions in opentrack's FSX options dialog. You must install SimConnect.msi and Visual C++ redist so that the library works. They're located in P3D's redist directory.
- - - - - 

Per <<https://www.prepar3d.com/forum/viewtopic.php?t=125109#p159985>>:
> Then, depending on the Simconnect version you used, the FSX Simconnect client will have an additional dependency on the VC++ 2005 runtime libraries, either the RTM version or the SP1 version, so they must be installed too, and since this is a 32 bit app, the x86 version of the runtimes must be installed, and be the correct one.

Per <<https://www.prepar3d.com/forum/viewtopic.php?t=125109#p161368>>:
> LM has said that they will have a look at the issue with RTM ans SP1. I hope they will find the problem if any.

Also, SimConnect version numbers in opentrack are all wrong. But for now let's test this hypothesis, shall we?

sthalik, 2017-07-07:
Try today's (July 7th) build. It works for me with the `redist/Interface/FSX-SP2-XPACK` SDK installed.

sthalik, 2017-07-07:
That is -- you MUST install the aforementioned `.msi` from P3D's install directory. As per the quoted forum posts, installing other `.msi`'s for P3D won't work.

PH-LCS, 2017-07-07:
Is this a newer version of the msi then the one came with the first install of p3dv4?

Verzonden vanaf mijn Samsung-apparaat

sthalik, 2017-07-07:
The version's 4.0.28.21686.

In any case you should have it in the "redist" directory.

PH-LCS, 2017-07-10:
Hi Stanislaw,

 

Removed all simconnects, install the latest .msi from version 4.0.28.21686 and my Opentrack still will not work.

I am getty crazy ……

 

Gr.

Jos

sthalik, 2017-07-10:
@PH-LCS did you download the snapshot from <<http://opentrack.misaki.pl>>?

PH-LCS, 2017-07-10:
Now I did and I am happy, it is working again.

Will have to adjust the settings, but that will be ok ….

 

Thanx man!

 

Gr.

Jos

sthalik, 2017-07-10:
FWIW, the old code worked, but individual version numbers were mislabelled. They existed but other the wrong names.

closed 2017-07-16 by sthalik

MathijsG, 2018-06-21:
I also encounter this problem, it's a little cryptic which version to pick.

sthalik, 2018-06-21:
@MathijsG,

1. Use the unstable version. The stable version has simconnect fully broken.
2. Pick versions until it doesn't error out anymore.