opentrack/opentrack #628: build 2.3.7

Forsaken-Player, 2017-06-05:
When the Mapping Properties "Curve goes negative", cursor doesn't stay at 0, y tries to follow the curve but with a positive sign.

[image]

sthalik, 2017-06-06:
Thank you @Forsaken-Player @Len62. I put up a build at http://opentrack.misaki.pl.

I'm thinking of replacing current Catmull-Rom splines with "monotonic cubic" splines that avoid these weirdnesses. But it's another bit of work.

Thought of catching some more Z's but Kropek was meowing incredibly loud this night.

sthalik, 2017-06-06:
[image]

See that bulge around X=0.4. That's how Catmull-Rom often performs also. I hate that too.

closed 2017-06-06 by sthalik

Len62, 2017-06-06:
The build you provided works fine. No more negative movement here.

sthalik, 2017-06-06:
:+1: I'll leave the issue open to investigate monotonic splines.

Forsaken-Player, 2017-06-07:
Same here, new build doesn't have the issue. I had no time to test it in games yet. I'm a bit busy at work this week, sorry on that.
Thank you for this great software!

commit a1ee9c7a (2017-06-06): spline: prevent returning absolute of negative value

Fixes: #628