opentrack/opentrack #629: No video problem

lowkeygit, 2017-06-06:
I plan to use Opentrack with Android phone for head tracking. I followed the instruction to set up: https://github.com/opentrack/opentrack/wiki/Smartphone-Headtracking
However, when I click the button "start", it shows no video.

I'm sure that the phone and PC are well connected. I am using samsung gt i9100. I have tried up my thought but cannot find a solution. What potential problem might be?

Thanks,

sthalik, 2017-06-06:
Not all trackers show any video. Android head tracking uses only inertial sensors and no camera.

closed 2017-06-06 by sthalik

lowkeygit, 2017-06-07:
May I use this setup if I want to use the smartphone for tracking the head movement and then collect the data. I mean, even if I cannot see the video, can I still collect the data successfully?

https://github.com/opentrack/opentrack/wiki/Smartphone-Headtracking

Thanks,

sthalik, 2017-06-07:
Yes. Use the .csv data logger in the options menu.

lowkeygit, 2017-06-07:
Thank you very much. I will try it.