opentrack/opentrack #630: Should mouse emulation not work on any game.

Muffasa, 2017-06-06:
Hi everyone, Correct me please if i am wrong but does mouse emulation not mean that any game with mouse look should work out of the box, I have been trying now for a couple of days to make the Forest work in just this way but alas have came up with nothing that works, mouse will operate when on desktop and while in the games menu, but not in game, which is a shame because the mouse emulation is the smoothest method i have tried with my Edtracker device, I have tried using the Vjoy method but it does not work correctly because it only moves the camera in one direction or the other.

looking forward to some ideas guys, thanks in advance.

sthalik, 2017-06-07:
Are you running the game as administrator? Don't.

Muffasa, 2017-06-07:
sthalik, yes actually i am, i am running everything as admin, will try it without and report back.

Muffasa, 2017-06-07:
Ok i have changed from run as admin on steam and the forest, that has not worked but it is late here so heading to bed, will try running Opentrack tomorrow without admin, see if that does anything, Thanks for your help.

MathijsG, 2017-06-07:
Sometimes you need to change the mouse behavior in-game to RAW input instead. Don't know how many games let you choose this, but this at least applies to GTA V when you want to use mouse based headtracking.

Muffasa, 2017-06-07:
MathijsG, yes thanks i read that somewhere, but i cant find this in the forest, i dont even think it has a config file that you can edit, If anyone knows where the config file is hiding i would love to hear from them,

sthalik, i have taken away all run as admin for all software now, and still the same result,

Thanks guys for your help its nice to be getting some ideas.

closed 2017-12-02 by sthalik