opentrack/opentrack #631: allow modules to call back into the app

sthalik, 2017-06-07:
I'm working on a way for modules (trackers, protocols, filters) to communicate with the main application. One pet peeve that motivates this is the inability of trackers to stop tracking after displaying an error message. Only protocols can do that.

Rather than going through all trackers and filters and implement the same way to cancel tracking, I think we should allow two-way communication for general purposes.

Presently the app calls into the modules to get required data, but modules can't ever communicate toward the app itself.

Thing that comes to mind is PointTracker sending LED positions so that the freetrack protocol can read it.

Another thing is sticking module configuration as tabs in the options menu. Creating a "dialog controller" would allow the modules to create their own keybindings, tabs, etc.

Can you think of things modules might want from the app? Let it rip!

Pinging @MathijsG, @Len62, @petsild.

MathijsG, 2017-06-07:
I only know of a 'simple' one, I always encounter: The clunky way of calibration right now, which needs to manually start the tracking. If I read you correctly, this could be differently after having this implemented?

Like for example a 'Start calibrating' button where the module itself starts the tracking process during calibration?

Len62, 2017-06-09:
Other then MathijsG's suggestion I can't think of anything else. Will do some more thinking about this and see if I can come up with anything.

closed 2018-08-03 by sthalik