opentrack/opentrack #632: WP port of opentrack no more available on store

kennyclaude, 2017-06-07:
hi would it be possible to help with this ?
can someone please contact the dev on this matter, can the app we brought back to store ?
was there an issue with the app being removed from store ?

sthalik, 2017-06-07:
No idea, I have no contact with @ZanderAdam. But you can build the app manually from <<https://github.com/ZanderAdam/OpenTrack.WindowsPhone>>.

sthalik, 2017-06-07:
There's a binary version at <<https://github.com/ZanderAdam/OpenTrack.WindowsPhone/releases>>. You might need to enable "sideloading apps" for that.

For the rest of the questions, no idea.

kennyclaude, 2017-06-07:
hi yes i know about the binary .appx file,
i tried all 3 ways of manual install on WP from SDcard, from local netportal, and from SDK appx deployment tool, and it didn't work, and gave all kinds of errors that i spent days on trying to solve and not succeeding,
i have a lumia 1520 + kinoni streamer + tridef - head tracking..

long stoory short opentrack udp sender is the only mean to getting a VR PC-aided experience,
and not having the required skills to build it myself i think i am just gonna way for it to be dropped again on store, 
i hope this happens thx all for bringing to us this magic..

ZanderAdam, 2017-06-07:
Looks like Microsoft removed the app because it never got an age rating. I'll have to contact them to resolve the issue. But you might be able to sideload in the mean time using one of the methods listen in the other issue: https://github.com/ZanderAdam/OpenTrack.WindowsPhone/issues/7

Please note that I do not own a windows phone anymore so only help I can give is what I find on the googles.

kennyclaude, 2017-06-07:
glad to see that there is still hope, WP scene is so tough and apps are so scarce seriously seeing glimmers of hope like this one really makes you see the real world more positively ,
i have 3 nokia lumia they are like the worst unhackable pieces of hardware lol

i am already on it trying to follow the microsoft tutorial
thx a lot Zander

sthalik, 2017-09-07:
Looks like fixed. Thanks @ZanderAdam!

closed 2017-09-07 by sthalik