opentrack/opentrack #633: Feature request: Opentrack tracking data output to SteamVR tracking protocol.

GrahamBotha, 2017-06-07:
Good day, is it possible to have opentrack tracking data sent via the SteamVR tracking protocol to SteamVR, so that it can be used as "native" tracking for Direct mode games?  This is exactly like receiving tracking data from a "base station" which would in this case be simulated by opentrack.  Having this functionality would allow any SteamVR game to receive 6 DOF opentrack data for any HMD in direct mode instead of using the gyro tracking that the devices already have.  Thus allowing us to use any open track compatible tracking device to do the tracking on behalf of the HMD in any supported direct mode VR game.
So in summary and I apologize for repeating this, but for some reason people on various forums tend to get it confused with what opentrack has already implemented vs this request.  The current version of opentrack uses SteamVR tracking data as input to allow the HMD to become the tracker for games in extended mode VR and simulate TrackIR\Opentrack input.  So opentrack data to go via the SteamVR protocol as output for direcmode VR would be a really great next feature, allowing opentrack to provide tracking on behalf of any non-mobile based HMD (I believe Riftcat does this for mobile based devices like the Samsung GearVR).  There is a fair community out there that don't have higher end VR HMD devices.  Having lower end units and IR-tracking clips would allow us accurate and cost effective 6 DOF tracking in direct mode, which we can only get via some integration with opentrack or directly from opentrack which would probably be the best solution.  

Thank you for indulging me and holding thumbs that you would consider this feature request.

Kind Regards

Graham Botha.

sthalik, 2017-06-07:
It's already been requested. See #494.

It's a lot of work for a hobby project though. Don't wait too eagerly however. It's probably gonna take many months till I get to it, if I'm the one to implement it.

Closing as duplicate of #494.

sthalik, 2017-06-07:
I can't think of a single gyro-based config that's worse than LEDs.

closed 2017-06-07 by sthalik

GrahamBotha, 2017-06-07:
Hi, thanks for the feedback and your awesome application. Yup it's a lot of
work for a one man band as a hobby project. I've also asked the Pimax 4k
engineers to give you a shout as they could potentially get the tracking
data from opentrack and pass it through their Piplay driver to steamVR.
Not sure if they will give you a shout though, but thank you very much for
opentrack.

Kind Regards

Graham Botha

GrahamBotha, 2017-06-07:
I don't know. Many of the mobile based hmd using cellphone gyros tend to
have yaw tracking drift. The Samsung gearVR is probably the exception. I
use  Pimax 4k hmd and they say it has a high quality gyro but the yaw drift
is quite bad. OpenTrack with the IR tracker totally fixed this for me in
Assetto Corsa which support both steam and opentrack protocols at the same
time. Getting the opentrack mapping to work correctly is maybe not a plug
and play solution but I found the led tracking to be very good within the
limitations of the camera :)

sthalik, 2017-06-07:
@GrahamBotha other VR software ought to include SteamVR drivers on their own. It's definitely outside the scope of the opentrack project to write HMD drivers for others.

As for gyro drift -- we solved it for the earliest Rift model by "yaw spring" code, just a few lines. Also magnetometers are very flaky with regard to speakers and headphones.

GrahamBotha, 2017-06-08:
@sthalik 
"Also magnetometers are very flaky with regard to speakers and headphones".  This makes total sense come to think of it when it comes to speaker and headphone magnetic interference.  Just a quick question though, as I am ignorant in this respect.  Does the "yaw spring code" manage to resolve the negative effect of headphone interference sufficiently or would that be most effective when external magnetic interference is minimal such as not using headphones or using smaller headsphones?  Just curious as a matter of interest.

Agreed that VR hardware ought to include SteamVR drivers on their own and you definitely are not responsible to write drivers for them.  I asked  Pimax to give you a shout in order to get the right information for integrating Opentrack data directly into their drivers from their side.  On the other hand should opentrack speak directly to SteamVR that would perhaps be the most ideal solution as anyone can then use opentrack for tracking purposes from a direct mode VR perspective.  But as you said its a lot of work and you already have a lot on your plate that you still need to get to.

All of that being said I don't want to take up too much of your time.  Thank you very much for opentrack as a whole, for the time spent in creating it and providing it to us.  It definitely solved my tracking issues within Assetto Corsa and it is an awesome piece of software.