opentrack/opentrack #635: AMD 17.6.1-June6 broke opentrack-PT tracker

albertolg, 2017-06-09:
after I installed latest AMD drivers opentrack stopped working, PS3 eye, randomly stop working, but if I open the Cl-eye-test then the camera is working and I can see the 3 dots. Reverted back to AMD 17.5.2 and opentrack is working again... both versions 2.3.6 and 2.3.7 have the issue... hope AMD will fix with next release or I can't update the drivers.

sthalik, 2017-06-09:
@albertolg 

Download `AmCap` from http://opentrack.misaki.pl and tell me if that works.

albertolg, 2017-06-10:
uummm... what should I see? I have 2 windows, a white one with a menu and nothing, the black one with numbers...

sthalik, 2017-06-10:
The timestamp increments and buffer size is more than zero, yes?

This is the WIP camera support code but I haven't removed the entire user interface yet.

albertolg, 2017-06-10:
yes, more than zero.. I'll try it again when my opentrack will froze... the problem was  the tracking frozing, than if I hit stop and start again it works again but randomly the problem occurs again... I will try reinstalling the new drivers, test and report back

sthalik, 2017-06-10:
What if you plugged the webcam into another USB slot? That makes the USB device's drivers install again.

albertolg, 2017-06-10:
haven't tried it yet, but what I noticed is that wen the problem happened I looked in opentrack and the dots in the monitor were stuck, but if I opened the CL-eye-test then the camera was working fine there but in opentrack was still stuck... don't you think if it was a problem with the drivers it could happen in every software I open? Reinstalled the new drivers, today I'll do more tests and if I can some videos...

albertolg, 2017-06-10:
update: tested for about 2 hours and not a single problem... I don't know what happened before... well, happy is working again

closed 2017-06-10 by albertolg

sthalik, 2017-06-10:
Your USB hu may be overloaded. Switch black/blue.

albertolg, 2017-06-10:
no hub... but as I said the camera was still working... could have been a problem of not rebooting after new drivers, I don't know, I just uninstalled and installed the older then did a reboot.. today I used DDU

sthalik, 2017-06-11:
What's DDU? Are you saying it works for you now?

albertolg, 2017-06-11:
DDU= display driver uninstaller, an utility to clean the system from old graphic drivers. Yes, now it's working fine, I don't know what happened, was so strange see the tracking freezing, this never happened to me all this time. Imagine a paused old video tape, where the scene jump from a frame ahead and one back in a loop, well, this is what I have seen when this happened. something like this but without the distortion https://www.youtube.com/watch?v=917isMQyvjw

albertolg, 2017-06-17:
just a little update.... the problems appeared again, also on 17.6.2, now I reverded back to 17.5.2 and will do more tests... this is really annoying

albertolg, 2017-06-17:
another update... the issue is also with 17.5.2, damn it....
@sthalik is possible to activate a log so I can see what is happening in my system?

sthalik, 2017-06-18:
@albertolg It's not the drivers. Switch the USB hub.

albertolg, 2017-06-18:
@sthalik probably win10 failed to install the drivers the first time (I had win8.1 then updated to w10), because I didn't had the icon to remove the hardware in the traybar (i have a ps3 eye), now I uninstalled and then reinstalled the drivers, changed USB port and I have again the icon. Tested for a couple of hours and didn't had any issue. Probably the cam where disconnected randomly causing to opentrack the problem.