opentrack/opentrack #639: camera offset + orientation axis bleed

tdoggo, 2017-06-22:
I have camera mounted +30deg pitch above my head for aruco tracking (so no jitter in "center"), with glyph mounted facing forwards (not at camera). I am using only aruco, no gyro/IMUs or anything so I don't have a fallback for orientation.

Camera offset setting +30deg removes the positional axis bleed in Y if I move in Z, but does not stop orientation bleeding as when I yaw my head - the output still bleeds into roll axis.

Zeroing the headset doesn't help as the zero position is fine but yaw still rolls, and adding a pitch offset to the game output data does not cancel it out. 

Seems like there is no compensation/cancelling for the camera position offset to calculate orientation, only position?

sthalik, 2017-06-22:
That's something we'd need to specifically calibrate for. My raw center position is (0, 10, 0). Pitch changes with yaw left/right.

We could calibrate for it but the procedure is necessarily more involved -- more code, more particular user actions.

sthalik, 2017-06-22:
I think that we can remove manual camera offset entirely, and replace it with offset based on camera center. Will publish a build sometime.

tdoggo, 2017-06-22:
is it that complicated?
If the camera offset angle is set correctly, it should just involve applying the inverse rotation before you  apply the raw orientation to the pose?

so that the camera recognises yaw in the 30deg plane as being "flat"

sthalik, 2017-06-22:
You'd need to calibrate for each axis, movement about the other two. For instance, how yaw affects roll and pitch is:

- roll some in both directions
- pitch some in both directions
- roll some, then pitch some, both in both directions (4x).

Similar for pitch and roll.

sniperwolfpk5, 2017-06-22:
I am using point tracker with FreeTrack I am facing similar problem. This feature would be nice and remove manual camera offsets. The automated offset will work much better than manual offsets.

tdoggo, 2017-06-22:
Oh yeah i see what you mean, if the glyph is not mounted exactly straight it becomes not simple

that could be caveat emptor / a requirement if you wish to use it in that way?

sthalik, 2017-06-22:
A covariance or correlation matrix would be appropriate here. I don't fully understand why the PT calibrator uses a precision matrix, though.

sniperwolfpk5, 2017-06-22:
I face this issue recently with new monitor, as my camera is above my head level. When I move forward/backward I have to adjust my head up/down for straight movement. Sorry for my bad English

tdoggo, 2017-06-22:
@sniperwolf positional problems are fixed by adjusting the "camera offset" settings, there is only a problem with turning your head

sthalik, 2017-06-22:
I removed manual "camera offset" for translation. Every time I tried using it, it wasn't right. It now centers varying upon the centered position. Here's a WIP build. Note that it can be broken in other places:
- I changed Accela gain and rotation nonlinearity meaning. The latter's now more useful and doesn't cause microstutter.
- I changed PT point extraction mechanism.
- I added additional PT stabilization mechanism.
- CPU usage is now 2% on my 4-core i7 with HT on. Previously it was 3-4%.
- Fixed applying RGB color weights when it's BGR that needs be used (it's exact opposite significance for R and B, omg).

NOTE that you can access debug data by opening opentrack from a text console. The debug info will then flow into that console window. You can open `Command Prompt`, drag opentrack.exe there and press Return.

Build: <<http://opentrack.misaki.pl>>. The filename is `opentrack-test-20170622_01.7z`.

I'm too sober to code anymore today. Anyone going for a Warka Strong?

:beer:   :octocat: :cat2: :beers:   :octopus:

My cat could eat a whole watermelon.

tdoggo, 2017-06-22:
much appreciated
its night time in middle earth but I will try it tomorrow and report back

albertolg, 2017-06-22:
@sthalik I tried your build, but there is a problem with the PT : if the tracking is lost (only 2 of 3 points visible) then the axis get crazy, then when it's tracking again it is reversed and act very strange

[image]

sthalik, 2017-12-21:
I'm working on interconnect calibration. Calibration works, but the data isn't used yet.

PoDBoT, 2018-01-28:
Hello. Camera offset settings are gone. But without it XYZ position is calculated totally wrong! For example: when I moving my head straight to the center of a monitor I expecting X and Y values would stay near 0, but in fact they changes as if I moving my head forward-down(camera is on monitor). The current variant could work correct only if camera were in center of a monitor.
Is there any way to fix this in the current version(2.3.9)???
[image]

sthalik, 2018-01-29:
I'll consider getting these settings back. Are there other examples where this setting is useful for you? Do you have any help with:

- yaw caused by pitch
- pitch caused by yaw
- yaw caused by X
- pitch caused by yaw+roll or similar

or any others?

sthalik, 2018-01-29:
Or other way of stating the same: Which of the axis influences another badly?

PoDBoT, 2018-01-29:
In my opinion, the problems you highlighted there are much less disturbing than the mismatch of movements along the axes. In fact, when I fine tuned the dimensions of the model and calibrated its position on the head, I did not even notice the problems with rotations. The only thing I noticed was pitch caused by yaw, but it was not so big to worry about it and could be attributed to my own "non-orthogonal" movements.

Maybe you should create something like "head-clip simulator" which will create video as if from a real camera but where all dimensions, positions and angles were precisely known. And then test opentrack with this video stream...

sthalik, 2018-01-31:
I wrote some working code for estimating axis interconnect. But it was too complex to use.

Basically we had `6*6 - 6` buttons to calibrate each axis in relation to another. I can't imagine users working with that kind of interface...

I'd need a list of axis where this is most significant. Having so many buttons for correlation isn't acceptable.

Usage was -- perform only movement around that axis. Will detect bleed causing movement along another axis.

Also geometry-wise, even if you perfectly center the camera, rotating around will cause bleed regardless. It's only the center pose that'll be free of bleed.

sthalik, 2018-01-31:
Now having that, we could have only 3 buttons (for yaw, pitch, and Z), estimating bleed over axis where it's the most visible. But don't expect it to remove bleed entirely. Also changing the sitting position, camera position or rotation, as well as clip orientation will require rerunning the calibration process entirely.

IslandJohn, 2022-12-23:
@sthalik do you have a commit where this was removed? I'd like to bring it back (happy to fork, or maybe make it an option) for use cases where the camera is mounted on a large display. Or is this supposed to be solvable with the current options? I'm finding that translation around Z is challenging.

sthalik, 2022-12-23:
@IslandJohn see https://github.com/opentrack/opentrack/blob/master/compat/correlation-calibrator.cpp

daphnisb, 2023-05-03:
> @sthalik do you have a commit where this was removed? I'd like to bring it back (happy to fork, or maybe make it an option) for use cases where the camera is mounted on a large display. Or is this [...]

I think it was removed in [2.3.8](https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3.8)
The 4th point on the list if I'm not mistaken.

I also got the Problem that i have the camera below my screen. It's a TV so there's also no possibility to mount it otherwise. I would really like this Feature to be Integrated again. Or an easier method would be to create a mix option where you can add Y axis movement according to the path traveled along the Z axis.

SkullCollector, 2024-02-22:
I would also like to chime in because I'm facing a bit of a headache.

My initial problem after finetuning every other setting to my liking was that roll causes X and Y translation. The model calibration did nothing to help, so I resorted to manual values and got a steady solution for roll, but now pitch and yaw are confounded.

Looking around and finding this issue makes me believe it's to do with my camera position: underneath the monitor angled up.

Is there a current setting I'm missing to accomplish the old camera offset?

sthalik, 2024-02-22:
Seems like we're gonna need the camera offset implemented again based on the amount of complaints.

YosypC, 2025-03-23:
I'm find the whole "Head center offset" concept very hard to understand. The automatic calibration yields very weird results, which don't seem to change the outcome. I still have Y input while moving the head perpendicular to the monitor, no matter what I input in the HCO, as my webcam is above it.

Maybe a clarification of this setting is needed since the offset angle is a been a missing feature for years and it doesn't seem it will come early. Does this function fix what the offset angle would accomplish?