opentrack/opentrack #643: Hot views together with Opentrack

XANoWAR10, 2017-06-30:
Hi all !
Sorry if i am wrong to post it here.
First of all want to thank an author of the Opentrack Project and those guys who help you. You are the best guys!
Next - are the 2 troubles for me using Opentrack...
1) Playing IL-2 BoS i set the hot views on the NumPad keys of the keyboard first.
Pressing them - you can turn to the saved field of view just immidiately.
It is saved. It works. But... When i launch Opentrack it stops working fully.
So i press that assigned button - but it doesn't move the view to that point. It only zooms a little bit in front and my view is close to the Sight...
Help pls. What should i do? I already reinstalled Opentrack (several versions) and also cleaned a registry before.
2) What is wrong with my settings when i see that the graph moves slightly down the X on the Opentrack?
Or even moves too high. And no possibility to catch it there...
Just imagine. We see 0 point on the diagramm. And then the graph moves down and then up.
So when i see it on the Pitch - it goes like you turn right - but slightly on the left in several miniseconds. And again right.

Waiting for your solutions.
And again Thank you
XANo

sthalik, 2017-06-30:
Regarding (2), I have little idea what's the problem at all. Please try describing it more clearly.

In any case, if you get unwanted position changes, disable Z for relative translation (or disable relative translation completely) and disable artificial neck.

If you haven't run the calibration procedure, run it.

Len62, 2017-06-30:
1. I just checked IL2-BOS and the snap-views using the numpad still worked for me with opentrack running.  Maybe check you key bindings in IL2? All programs running w/admin privileges?

@sthalik 

2. I think this was the issue that appeared with 2.3.7 (may be wrong) where there was some unintended negative movement in the mapping (graph lines dipped below on the horizontal plane).

XANoWAR10, 2017-06-30:
1. Yes - w/admin privs. Il-2 snapviews are working without opentrack. Some conflict maybe. IL-2 support said that the problem is in the mapping of Opentrack.

XANoWAR10, 2017-06-30:
Z and Roll are disabled btw

XANoWAR10, 2017-06-30:
1. The thing is that lots of my friends tell me how they do it. So i do the same. I even tried their config files. But still fail

Len62, 2017-06-30:
Since it's working for me it's hard to narrow it down. I'll have to spend some time thinking about it. What ver of OT are you currently using? What type of head tracking are you using? What type of joystick (i ask this because sometimes FFB sticks cause problems)? Try this version, dated 6/22/17: https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa?dl=0&lst=

XANoWAR10, 2017-06-30:
Thx Len62 for wasting your time... I appreciate it.

I installed the latest version yesterday. OT ver is 2.3.7
I use this tracker: https://www.avito.ru/moskva/tovary_dlya_kompyutera/movehead._luchshaya_klipsa_freetrack_trackir_778545450. 
It's Mr Sanchos's creation. Together with PS3 EYE.
I' use ThrustMaster Warthog.
Will use suggested version in 6 hours as soon as will get back home.
Again Thank You

XANoWAR10, 2017-06-30:
Regarding (2)... Answering Sthalik...
I use config.ini file that i created before. Also Yesterday i created new profile in 2.3.7 version.
But. What i saw - almost on every graph it was like this
0x0; -1x-1; 0x0; 4x10; 19,5x20
And when i try to rewrite the graph using mouse, even deleting some points, - still i have a minus position on the graph.

sthalik, 2017-06-30:
Use the build linked to by @Len62 in https://github.com/opentrack/opentrack/issues/643#issuecomment-312246687

Len62, 2017-06-30:
Not a waste of time, glad to try and help out.
"And still using i-16 it's pretty useful to have the snapviews on it's gauges."

Had not thought about that, especially with the I-16!

XANoWAR10, 2017-06-30:
well
with that version
graph is ok - without minus
but snapviews are still not working..
Thank you!

Len62, 2017-06-30:
I know it's a lot of work but can you re-set the IL2 snapviews to their defaults, and then check to see if the default settings work with OT running? If they do, then we may be able to narrow down the issue. I don't currently use the snapviews but I'll try and set some up to see what happens.

XANoWAR10, 2017-06-30:
ok - i'll do
will report here later for sure

XANoWAR10, 2017-06-30:
Fully dismissed snapviews and tried it out - fail
Whatever i am not asking aanymore - cause it's hard as i already understood
May be one day there will be the solution
Whatever... Thx guys

sthalik, 2017-07-01:
If you disable tracking completely, do snapviews work?

XANoWAR10, 2017-07-01:
Yep
it works without tracking

closed 2017-09-07 by sthalik