opentrack/opentrack #644: Loss of Calibration

Len62, 2017-07-04:
On the latest test build, dated 6/22/17, OT keeps losing the calibration setting. I had not noticed this earlier as I was only using it for short periods. Tonight I was on a longer session when I noticed the calibration issue. I'm going to try a completely new ini file and see what happens. Really strange.

sthalik, 2017-07-04:
I could use steps to reproduce. Thanks.

Len62, 2017-07-04:
If I yaw right 180 and pitch down 180 and hold that position after a couple of seconds the calibration goes off. Same going left and down. No overlap of LEDs occurs. I have to do some more testing. If I get a chance later today I may make a short video.

sthalik, 2017-07-04:
Note, calibration happens before mapping. It doesn't matter if it's 180/180, only the raw values matter. 

What do you mean by "go off"?

albertolg, 2017-07-04:
I confirm the issue, if you just cover 1 led you'll see what happens

albertolg, 2017-07-04:
https://user-images.githubusercontent.com/7831332/27453483-eac2e3c4-5796-11e7-9e21-992ebcf75919.png

albertolg, 2017-07-04:
[image]

sthalik, 2017-07-04:
I covered one LED. Didn't get a sensible pose or multiple samples. Didn't see any issues. What happens on your end?

albertolg, 2017-07-04:
I made a video

https://youtu.be/eDaaOgFXtTg

Len62, 2017-07-04:
Here's another vid: https://www.dropbox.com/s/j1hcox4ehivjarw/ot%20cal%20test.mp4?dl=0

"go off" just meant the calibration marker moved as you can see in the video. (I need to clarify better what I mean.)

Len62, 2017-07-05:
I tested 2.3.7 release and the 20170608-accela-min-delay.7z and both did not exhibit this problem.

albertolg, 2017-07-07:
tested opentrack-test-20170707_00.7z and the issue is still there

sthalik, 2017-07-07:
It's because of auto threshold. It won't happen with it disabled. I have the same issue.

sthalik, 2017-07-07:
Try today's build labelled `_01`. I think there was some faulty code making pose return all zeros.

sthalik, 2017-07-07:
@Len62 on the video, do you have dynamic pose enabled? You should always have it disabled for clips. It's only for caps.

albertolg, 2017-07-07:
@sthalik tested it, there is still a problem, now it is reset only after the dynamic pose timeout have elapsed

sthalik, 2017-07-07:
It's meant to hold the previous pose. The pose should never reset to zero because a timeout elapsed, etc. Are there circumstances where it resets to zeros?

sthalik, 2017-07-07:
I ninja updated the build. There's a separate deadzone for LEDs and it was too high. If you get microstutter or need high sensitivity, try it.

albertolg, 2017-07-07:
now it's working fine

Len62, 2017-07-07:
It was the dynamic pose. On the 6 22 build it still loses tracking with the extreme head movement but it does not lose calibration. On the 7 7 .01 build it no longer loses tracking or calibration.

closed 2018-05-14 by sthalik