opentrack/opentrack #645: UDP active getter

zakamik, 2017-07-04:
Is it possible to make reverse (to existing) UDP protocol, where one could specify destination host and port for UDP? This might come more suitable for wireless head trackers.

sthalik, 2017-07-04:
Yes. There's already UDP for both the sender and the receiver.

zakamik, 2017-07-04:
That's great news but I could not find IP address configuration in current release...

sthalik, 2017-07-04:
The "tracker" receives data while listening on all addreses on the given port. The "protocol" sends data to the specified address.

zakamik, 2017-07-04:
I am not sure if I get your description properly, as my copy starts localized without option to turn back to the English interface.
The idea is to be able to configure UDP receiver so that it would connect to the tracker' (physical device) IP similar way as it is done for the COM port in hatire plugin.

sthalik, 2017-07-05:
@zakamik there's an option to disable translation in the options dialog. It has a British flag next to it.

> The idea is to be able to configure UDP receiver so that it would connect to the tracker' (physical device) IP similar way as it is done for the COM port in hatire plugin.

On the device that receives pose data, use the udp _protocol_. On the target device that needs the data, use the udp _tracker_.

```
[Source] -> [Filter] -> [Sink]

Source ≡ Tracker
Sink ≡ Protocol
```

zakamik, 2017-07-05:
Thank you for the hint with translation. I suspect that we are talking about different matters. Let me rephrase.
Suppose that I have a PC with Opentrack running on it, and head tracker with wifi, running inertial tracking firmware [TR]. TR has its own IP address with UDP port opened for listening. When someone connects to this port, TR starts to send head position data via that connection and does this until connection is closed. I would like to configure Opentrack and specify TR's IP and port so that when Opentrack is started, it connects to TR to get position data.

sthalik, 2017-07-05:
1) You'd need another app to perform the connection. You also need data format for "TR" to transcribe into opentrack's format.

This can be done in ~20 lines of Python.

2) UDP connections aren't "closed". They're stateless. Only via ICMP the receiver can notify nothing's listening for the data but that can't be taken for granted.

3) Send me the documentation on the UDP specifics. Maybe I'll be able to help you.

zakamik, 2017-07-05:
Basically, the plan was to have a direct connection initiated by opentrack without any extra apps. UDP format for TR is the same as 'UDP over network'. It works, but changing IP in firmware seems less convenient, than in PC app.

sthalik, 2017-07-05:
```python
#!/usr/bin/env python2

from time import sleep
from sys import argv, exit
from socket import socket, AF_INET, SOCK_DGRAM, SOL_SOCKET, SO_REUSEADDR
from struct import pack, unpack
import os

if os.name != 'nt':
    raise "only for Windows socket semantics"

# you can use the "pack" function
# cf. <https://docs.python.org/2/library/struct.html#format-characters>
YOUR_MESSAGE_TO_SEND = 'something'

# same as opentrack's port
OUR_PORT = int(argv[3])

# the sender's udp bind address
TARGET_HOSTNAME = argv[1]
TARGET_PORT = int(argv[2])

sock = socket(AF_INET, SOCK_DGRAM)
sock.setsockopt(SOL_SOCKET, SO_REUSEADDR, 1)
sock.bind(("0.0.0.0", OUR_PORT))
for _ in xrange(1, 5):
    sock.sendto(YOUR_MESSAGE_TO_SEND, (TARGET_HOSTNAME, TARGET_PORT))
    sleep(.016)
sock.close()
```

Run as:

> otr-initiate-udp.py google.com 666 4242

Where `google.com` is the networked sender's address (can be IP, can be hostname), `666` is the port to send the activation request to, and `4242` is opentrack's listen port.

You still need to edit `YOUR_MESSAGE_TO_SEND` to the activation request.

I verified that both programs can listen on the same `(source-address, port)` tuple at the same time.

zakamik, 2017-07-05:
Thank you for the script, will try and let you know.

closed 2017-09-07 by sthalik