opentrack/opentrack #646: Not all keys on keyboard can be assign to functions

BA1-Templar, 2017-07-06:
Hello,
I want to report bug:
I'm trying to assing key to _Zero while held_, and I can't assign a lot of keys, like alt, ctrl, tab, caps-lock etc,
Or maybe there is some other way, like writing key into profile ini?

Len62, 2017-07-06:
alt and ctrl are just modifiers, you must press another key/joystick button while holding down alt or ctrl. I don't think tab and caps-lock work.

sthalik, 2017-07-06:
It's a design decision that modifier keys can't be bound.  But you're right about the Tab and Caps Lock keys. They weren't included by accident.

BA1-Templar, 2017-07-06:
Ok, thanks for info.
Pozdrawiam!

sthalik, 2017-07-06:
@BA1-Templar ściągnij dzisiejszy z http://opentrack.misaki.pl. Może będziesz miał jeszcze jakieś uwagi.

BA1-Templar, 2017-07-06:
Ok, pobrany i zainstalowany.
Pomęczę tą wersję i jak coś znajdę to dam znać.

BA1-Templar, 2017-07-07:
Mam kolejny pomysł odnośnie klawiszy.
Chodzi o dodanie do możliwych klawiszy myszki. Jak i PPM oraz LPM jak i przycisków przód/tył na myszce.

sthalik, 2017-07-07:
To trzeba by zrobić kod osobno od joysticka i klawiatury. Przemapuj sobie jakimś programem, tak będzie łatwiej.

closed 2017-09-07 by sthalik