opentrack/opentrack #648: S2Bot as input source

achipa, 2017-07-07:
I've written a little hack that allows users to use OpenTrack via S2Bot*. It's implemented as an extra input source, what would be the right way of upstreaming it from by fork branch (https://github.com/achipa/opentrack/tree/s2bot/tracker-s2bot) ?

*S2Bot is a cross-platform solution originally to link between sensor-enabled hobby devices like LEGO WeDo/Mindstorms, BBC Micro:bit and Scratch (http://www.picaxe.com/Teaching/Other-Software/Scratch-Helper-Apps/). I'm using the same protocol normally Scratch would use to connect (closest probably to freepie-udp from the existing ones).

closed 2017-09-07 by sthalik

StoryLabTAMU, 2019-07-10:
Hi achipa, I just used OpenTrack and opened S2Bot as input, but there's no selection to pair which device, and I assume the S2Bot in here is native not App version, right? So how do I connect a Sensortag 2 to it?
Thanks in advance!

achipa, 2019-07-15:
@StoryLabTAMU This _is_ for the S2Bot App - you should connect the Sensortag 2 in exactly the same way you would for Scratch. The OpenTrack part of this solution doesn't really care how (or which) device is connected to S2Bot. http://www.picaxe.com/docs/s2bot_app.pdf should be a good starting point on how to connect actual physical devices to S2Bot.

Currently there is no (good) way of _simultaneously_ connecting multiple devices through S2Bot, if that's the question.