opentrack/opentrack #651: Mapping issue - stuttering when in dead zones

unhappytroll, 2017-07-11:
[image]
This is how my mapping window look like as of now - curves wasn't so broken in previous version. 
also I have a video of how this strange mapping now behaves:
https://youtu.be/kNZH770XoYk

this was not in previous version too. And if I remove dead zones this stuttering in them is not happens.
what is that and how I can avoid it (except rollbacking into previous version)?

Len62, 2017-07-11:
Try this version: https://www.dropbox.com/sh/823dd4wyhixv5a7/AABIY7dkHAjCgkVsS6mrVQxEa/opentrack-test-20170622_01.7z?dl=0

It fixes the mapping issue.

unhappytroll, 2017-07-11:
oh. ok, thanks. 
Yes, it fixed it. Thanks again.

closed 2017-07-11 by unhappytroll

Len62, 2017-07-11:
You're welcome!

sthalik, 2017-07-15:
I really need to release 2.3.8. I can't for the moment.

@Len62 Are there any new issues in the nightly build?

Len62, 2017-07-15:
I haven't run into any issues with the 20170707-01 build. I've been using it since you posted it.